DSheirer/sdrtrunk #1243: Icecast generating too much silence

doug-hoffman, 2022-05-07:
We're currently generating too much silence for icecast. This can be observed with clients that indicate playback position and/or buffer size, such as mpv:

After 5 seconds of playback:
```
$ mpv http://10.1.1.25:8000/test1
 (+) Audio --aid=1 (mp3 1ch 8000Hz)
File tags:
 icy-title: Scanning...
AO: [coreaudio] 8000Hz mono 1ch floatp
A: 00:00:05 / 00:01:09 (8%) Cache: 63s/513KB
```

After 2 minutes, the buffer has grown by 20 seconds:
```
$ mpv http://10.1.1.25:8000/test1
 (+) Audio --aid=1 (mp3 1ch 8000Hz)
File tags:
 icy-title: Scanning...
AO: [coreaudio] 8000Hz mono 1ch floatp
A: 00:02:00 / 00:03:23 (59%) Cache: 82s/664KB
```

This issue has also been observed while streaming with XiiaLive on Android as the playhead slowly drifting behind the most recent audio:

[image]

doug-hoffman, 2022-05-07:
Adding debug:

```
                    {
                        List<byte[]> silenceFrames = mSilenceGenerator.generate(PROCESSOR_RUN_INTERVAL_MS);
                        int tot = 0;
                        for(byte[] silence: silenceFrames)
                        {
                            tot += silence.length;
                            broadcastAudio(silence, null);
                        }
                        mLog.debug("Silence bytes:" + tot);
                    }
```

We're sending 2,304 bytes/sec (18,432 bits/sec) of silence for 8kHz @ 16kbps:

```
16:59:06.987 DEBUG i.g.d.a.b.AudioStreamingBroadcaster - Silence bytes:2304  [41MB/80MB 51%]
16:59:07.984 DEBUG i.g.d.a.b.AudioStreamingBroadcaster - Silence bytes:2304  [42MB/80MB 52%]
16:59:08.980 DEBUG i.g.d.a.b.AudioStreamingBroadcaster - Silence bytes:2304  [43MB/80MB 53%]
16:59:09.980 DEBUG i.g.d.a.b.AudioStreamingBroadcaster - Silence bytes:2304  [44MB/80MB 55%]
16:59:10.977 DEBUG i.g.d.a.b.AudioStreamingBroadcaster - Silence bytes:2304  [44MB/80MB 56%]
```

doug-hoffman, 2022-05-08:
Given the need to cope with VBR, targeting a specific delivery bitrate is out. Handling MP3 audio at the frame level seems like the next best option to get consistent delivery. A lightly-tested example for silence generation is in #1244.

It seems wise to extend this to the entirety of audio we broadcast in ```RecordingQueueProcessor```, for even better results. My understanding of MP3 encoding is that there is _some_ extra audio at the beginning and end of each file, so we're likely sending a little extra with actual recordings that we could offset occasionally by dropping an extra frame or two of silence.

zacharyfleck, 2022-09-20:
@DSheirer I see this issue referenced back in May. I recently got reports of this issue happening on my streams (Think I'm on 0.5.0 Beta 3, would have to check when I'm home from work.) and was Wondering if the commit referenced (https://github.com/DSheirer/sdrtrunk/commit/8bfb1d54702ce89f44094a76346e98f4b9f7e255) includes the fix as #1243 currently remains open.

doug-hoffman, 2022-09-21:
I don't believe this has been fixed. I've been running with #1244 since mid-May with no issues, streaming to both BCFY and my personal icecast server.

zacharyfleck, 2022-09-21:
@doug-hoffman Thanks for the info. I'll probably pull those changes today and give it a try.

doug-hoffman, 2022-12-18:
Resolved by #1244.

closed 2022-12-18 by doug-hoffman