DSheirer/sdrtrunk #1249: Airspy issue in 0.5.0-beta3

davidbelle, 2022-05-11:
Hi all

I have compiled the latest commit (1243) but a few of my machines with Airspy tuners will crash with a very lovely memory access violation.  Crash isn't immediate, could last a while before crashing. 

Has anyone else experienced this?

[memoryviolation.txt](https://github.com/DSheirer/sdrtrunk/files/8666686/memoryviolation.txt)

Brad-git-man, 2022-05-11:
Yeah, I run five 2.8MHz SDRs all at once and after maybe around 2 hrs I get a memory leak. I also get other errors like can't play a channel that it was playing just fine earlier.

srsanford, 2022-05-25:
I have had similar results including after compiling #1253 .

18:48:06.633 INFO  i.g.d.m.d.p.a.P25P1AudioModule - P25P1AudioModule patch - mEncryptedCall is true  [611MB/1GB 55%]
18:51:51.137 ERROR i.g.d.s.t.u.USBTunerController - Error submitting USB transfer - status [-1] LIBUSB_ERROR_IO - previous status [1] and transferred [0]  [191MB/1GB 17%]
18:51:51.142 INFO  i.g.d.s.t.m.DiscoveredTuner - Tuner Error - Stopping - Airspy USB Bus:1 Port:1 Error: Usb I/O Error - Can't submit buffers to tuner - stopping tuner  [197MB/1GB 18%]
18:51:51.147 INFO  i.g.d.s.t.m.DiscoveredTuner - Stopping Tuner: Airspy USB Bus:1 Port:1  [205MB/1GB 18%]
18:51:51.151 INFO  i.g.d.m.d.p.P25TrafficChannelManager - Stopping traffic channel: West Hartford_West Hartford_TRAFFIC_33  [206MB/1GB 18%]
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x00007ff890ab3416, pid=13196, tid=6832
#
# JRE version: OpenJDK Runtime Environment (17.0.2+9) (build 17.0.2+9-LTS)
# Java VM: OpenJDK 64-Bit Server VM (17.0.2+9-LTS, mixed mode, sharing, tiered, compressed oops, compressed class ptrs, g1 gc, windows-amd64)
# Problematic frame:
# C  [ntdll.dll+0x63416]
#
# No core dump will be written. Minidumps are not enabled by default on client versions of Windows
#
# An error report file with more information is saved as:
# C:\srs\sdr-trunk-0.5.0-beta3-may24\bin\hs_err_pid13196.log
#
# If you would like to submit a bug report, please visit:
#   https://bell-sw.com/support
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
[hs_err_pid13196.log](https://github.com/DSheirer/sdrtrunk/files/8770969/hs_err_pid13196.log)

closed 2026-11-06 by DSheirer