DSheirer/sdrtrunk #1253: USB Buffer Transfer Fails On Resubmit

DSheirer, 2022-05-17:
Some windows users are reporting that their USB tuners are generating an error when streaming and attempting to resubmit a USB transfer buffer to libusb.

On submitting the transfer, libusb reports ERROR_BUSY and according to the javadoc this indicates that libusb considers the transfer as 'already submitted'.  This is odd because the transfer was reported as completed by libusb during the handle events timeout invocation, so libusb has told us that it's finished with the transfer.

When this happens:
* transfer.status() reports: LibUsb.TRANSFER_COMPLETED (0)
* transfer.actualLength() reports: zero bytes
* on resubmit, libusb reports: LibUsb.ERROR_BUSY

This seems like a bug in libusb because libusb is telling us that the transfer is complete but then also telling us that the transfer is still in use when we attempt to resubmit it.

Experiment to see if we can ignore the completed timeout and allow libusb an opportunity to fill the buffer again.  However, we run the risk that libusb will not use this transfer, and potentially exhaust all in-progress transfers to the point that data no longer streams.

DSheirer, 2022-05-24:
Closed with PR.

closed 2022-05-24 by DSheirer

jdelhome3578, 2022-05-25:
@DSheirer this does not seem to be an issue in 0.5.0 beta 2

DSheirer, 2022-05-25:
@jdelhome3578 This issue came about with changes introduced between beta 2 and beta 3.