DSheirer/sdrtrunk #1260: DMR MS always sync lost

edajimak, 2022-05-23:
I noticed that when receiving DMR voice frame from a mobile station (MS), almost all packets are lost. The current implementation of DMR decoder use Costas Loop (PLL) and symbol interrelation with timing correction. This circuit has inertia and requires a continuous carrier signal to adjust phase error and timing error.

So the DMR decoder lost synchronization on each 30ms MS burst, because the signal from the MS does not have a continuous carrier and CACH
[image]

I modeled how the initial parameters (init position of symbol) of the InterpolatingSampleBuffer affect the demodulation result.
The figure below shows the FM demodulated DMR signal, the dots indicate the samples for the current symbol which returns the InterpolatingSampleBuffer
[image]
As you can see, the start position fails to synchronize for a short burst and demodulator get a wrong result

Nokoa, 2022-12-01:
I noticed that during the early dev of DMR for SDRTrunk perhaps MS worked, based on the original issue for its creation #6 . I wonder what has broken it since?

DSheirer, 2026-11-08:
This may be resolved with the latest updates to the DMR decoder.  Please advise if it's not.

closed 2026-11-08 by DSheirer