DSheirer/sdrtrunk #1268: 0.5.0-beta3 sync loss 3-4 min after start up consistently

gVkWY8NJAa, 2022-06-03:
I am not sure how to describe the actual issue other than on win10 running 0.5.0-beta3 with an rtl-sdr v3 I will have a flood of sync loss messages with what appears to be random codes appended to the message ie no discernible pattern to the numbers. During this flood audio appears to no longer be able to be decoded. 

Restarting the application will result in normal operation again for the first 3-4 min until the next flood of sync loss messages and no audio.

I thought maybe the freq was drifting as the sdr heats up, however **reverting back to 0.5.0-beta2 seems to resolve this issue.**

Please let me know if any log info would be desirable (which logs and how to grab them).

DSheirer, 2022-06-04:
There was a change between beta2 and beta3 where the tuner's PPM setting was reset to default.  Can you check the PPM setting of your tuners to ensure the signal is aligned to the frequency?

w8jfj, 2022-06-06:
[gVkWY8NJAa](https://github.com/gVkWY8NJAa)

I followed Denny's suggestion and assured my PPM was setup properly. 

Also, look at the Channel tab in the Now Playing section. Assure you have a strong enough signal.  I set mine on the Tuner's tab as mode linearity and had to set the Master at 20. This set my signal level at about -29db. However I was still getting Sync Loss messages and losing voice shortly after startup while running in Polyphase mode on Tuner Channel. Over the weekend I ran successfully for over 48 hours running in Hetrodyne mode. You might try that and see what happens. 

This afternoon I tried Polyphase again and it started failing within in a half hour after startup. 

There is also discussion about this at https://groups.google.com/g/sdrtrunk/c/uJaxqcux52A.

Good Luck, John.

closed 2026-11-07 by DSheirer