DSheirer/sdrtrunk #1269: DMR Cap+ No Talkgroup IDs, or RID shown

Nokoa, 2022-06-04:
SDRTrunk will usually not display the talkgroup ids for group voice calls in this Motorola Cap+. I looked at event logs, and I am seeing voice headers with the ID, and terminators with the ID. This is on a multiple frequency Motorola Cap+ system.

Nokoa, 2022-06-05:
This occurs on systems where the group call type is TXI. Checked on another motorola cap + where regular group voice calls are used, and talkgroup ids, and radio ids, do appear.

Nokoa, 2022-06-05:
```
DECODED Message Logger

20220605 094020,PASSED,<-> SYNC LOSS - BITS PROCESSED [9600]
20220605 094022,PASSED,<-> SYNC LOSS - BITS PROCESSED [9600]
20220605 094021,PASSED,<-> SYNC LOSS - BITS PROCESSED [2280]
20220605 094021,PASSED,<-> SYNC LOSS - BITS PROCESSED [288]
20220605 094021,FAILED,[CRC-ERROR] CC:2 CSBK *UNKNOWN* MOTOROLA CAP+ UNKNOWN OPCODE:40 MSG:A810A4200400000000002468
20220605 094021,PASSED,<-> SYNC LOSS - BITS PROCESSED [288]
20220605 094021,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:5 LCN:3 MSG:BE10E4201400000000007478
20220605 094021,PASSED,<-> SYNC LOSS - BITS PROCESSED [288]
20220605 094021,PASSED,<-> SYNC LOSS - BITS PROCESSED [288]
20220605 094021,PASSED,<-> SYNC LOSS - BITS PROCESSED [288]
20220605 094021,PASSED,<-> SYNC LOSS - BITS PROCESSED [288]
20220605 094021,PASSED,     BS VOICE A
20220605 094021,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:5 LCN:3 MSG:BE10E4201400000000007478
20220605 094021,PASSED,<-> SYNC LOSS - BITS PROCESSED [288]
20220605 094021,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:5 LCN:3 MSG:BE10E4201400000000007478
20220605 094021,PASSED,<-> SYNC LOSS - BITS PROCESSED [288]
20220605 094021,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:5 LCN:3 MSG:BE10E4201400000000007478
20220605 094021,PASSED,<-> SYNC LOSS - BITS PROCESSED [288]
20220605 094021,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:5 LCN:3 MSG:BE10E4201400000000007478
20220605 094021,PASSED,<-> SYNC LOSS - BITS PROCESSED [288]
20220605 094021,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:5 LCN:3 MSG:BE10E4201400000000007478
20220605 094021,PASSED,<-> SYNC LOSS - BITS PROCESSED [288]
20220605 094021,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:5 LCN:3 MSG:BE10E4201400000000007478
20220605 094021,PASSED,     BS VOICE A
20220605 094021,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:5 LCN:3 MSG:BE10E4201400000000007478
20220605 094021,PASSED,SLC MOTOROLA CAP+ SITE:1 REST CHANNEL:LSN:5 LCN:3 MSG:F1004081C
20220605 094021,PASSED,CC:2 BS VOICE B
20220605 094021,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:5 LCN:3 MSG:BE10E4201400000000007478
20220605 094022,PASSED,CC:2 BS VOICE C
20220605 094022,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:5 LCN:3 MSG:BE10E4201400000000007478
20220605 094022,PASSED,SLC MOTOROLA CAP+ SITE:1 REST CHANNEL:LSN:5 LCN:3 MSG:F1004081C
20220605 094022,PASSED,CC:2 BS VOICE D
20220605 094022,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:5 LCN:3 MSG:BE10E4201400000000007478
20220605 094022,PASSED,FLC UNKNOWN OPCODE:32 VENDOR:MOTOROLA CAP+ MSG:2010700000140042AB00
20220605 094022,PASSED,CC:2 BS VOICE E
20220605 094022,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:5 LCN:3 MSG:BE10E4201400000000007478
20220605 094022,PASSED,SLC MOTOROLA CAP+ SITE:1 REST CHANNEL:LSN:5 LCN:3 MSG:F1004081C
20220605 094022,PASSED,CC:2 BS VOICE F
20220605 094022,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:5 LCN:3 MSG:BE10E4201400000000007478
20220605 094022,PASSED,     BS VOICE A
20220605 094022,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:5 LCN:3 MSG:BE10E4201400000000007478
20220605 094022,PASSED,SLC MOTOROLA CAP+ SITE:1 REST CHANNEL:LSN:5 LCN:3 MSG:F1004081C
20220605 094022,PASSED,CC:2 BS VOICE B
20220605 094022,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:5 LCN:3 MSG:BE10E4201400000000007478
20220605 094022,PASSED,CC:2 BS VOICE C
20220605 094022,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:5 LCN:3 MSG:BE10E4201400000000007478
20220605 094022,PASSED,SLC MOTOROLA CAP+ SITE:1 REST CHANNEL:LSN:5 LCN:3 MSG:F1004081C
20220605 094022,PASSED,CC:2 BS VOICE D
20220605 094022,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:5 LCN:3 MSG:BE10E4201400000000007478
20220605 094022,PASSED,FLC UNKNOWN OPCODE:32 VENDOR:MOTOROLA CAP+ MSG:2010600000140042AB00
20220605 094022,PASSED,CC:2 BS VOICE E
20220605 094022,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:5 LCN:3 MSG:BE10E4201400000000007478
20220605 094022,PASSED,SLC MOTOROLA CAP+ SITE:1 REST CHANNEL:LSN:5 LCN:3 MSG:F1004081C
20220605 094022,PASSED,CC:2 BS VOICE F
20220605 094022,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:5 LCN:3 MSG:BE10E4201400000000007478
20220605 094022,PASSED,     BS VOICE A
20220605 094022,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:5 LCN:3 MSG:BE10E4201400000000007478
20220605 094022,PASSED,SLC MOTOROLA CAP+ SITE:1 REST CHANNEL:LSN:5 LCN:3 MSG:F1004081C
20220605 094022,PASSED,CC:2 BS VOICE B
20220605 094022,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:5 LCN:3 MSG:BE10E4201400000000007478
20220605 094022,PASSED,CC:2 BS VOICE C
20220605 094022,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:5 LCN:3 MSG:BE10E4201400000000007478
20220605 094022,PASSED,SLC MOTOROLA CAP+ SITE:1 REST CHANNEL:LSN:5 LCN:3 MSG:F1004081C
20220605 094022,PASSED,CC:2 BS VOICE D
20220605 094022,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:5 LCN:3 MSG:BE10E4201400000000007478
20220605 094022,PASSED,FLC UNKNOWN OPCODE:32 VENDOR:MOTOROLA CAP+ MSG:2010600000140042AB00
20220605 094022,PASSED,CC:2 BS VOICE E
[... 407 lines cut ...]
20220605 094032,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:2 LCN:1 MSG:BE10C100000000000000357D
20220605 094032,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:2 LCN:1 MSG:BE10E1000000000000004E15
20220605 094032,PASSED,SLC MOTOROLA CAP+ SITE:1 REST CHANNEL:LSN:2 LCN:1 MSG:F1001085D
[... cut ...]
20220605 094033,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:2 LCN:1 MSG:BE10E1000000000000004E15
20220605 094033,PASSED,SLC MOTOROLA CAP+ SITE:1 REST CHANNEL:LSN:2 LCN:1 MSG:F1001085D
20220605 094033,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:2 LCN:1 MSG:BE10C100000000000000357D
20220605 094033,PASSED,CC:2 MOTOROLA CAP+ SYSTEM STATUS [--] REST LSN:2 LCN:1 MSG:BE10E1000000000000004E15
20220605 094033,PASSED,<-> SYNC LOSS - BITS PROCESSED [288]
20220605 094033,PASSED,<-> SYNC LOSS - BITS PROCESSED [288]
20220605 094035,PASSED,<-> SYNC LOSS - BITS PROCESSED [9600]

```

Nokoa, 2022-06-06:
Messed around with the code a bit, and I noticed that some LC messages were checking for their validity. However because I believe this system is RAS. Not all validly had the or clause for the ignore crc check. I added one for the voice header processing. 

Please also note that originally here I mentioned that tgid and rid are not shown, but for this particular system I had the ignore crc off. Which resulted in the talkgroups not showing up. However after turning it on, while TGID did show, RID did not. Once I added that crc check ignore clause, they once again showed up.

Still isn't working flawlessly though. After the call ends, the TGID stays on the active channel(data). Also there are still times when the TGID is not shown immediately or at all.

kazzaw, 2022-11-14:
Apologies for bumping an old issue, but I also see this issue as well with a system using RAS. 

It isn't a CAP+ system and only SDRTrunk sees it as a Cap+ system (DSDPlus sees it as a standard DMR 1 frequency system), however I believe thats a different issue from what is being discussed here. 

The system is active with calls and I can see/hear them when using a scanner or other software, but SDRTrunk won't identify them, play them or recall them.

Nokoa, 2022-11-14:
Do you have ignore crc on in the system configurations? I believe that when RAS is active CRC checks fail for certain packets without the RAS key loaded into the radio, radios, scanners, SDRTrunk, and other programs think the packets are not valid, as intended. So they have to ignore the results of the CRC checks in order to continue processing them. On SDRTrunk you have to enable ignore CRC Checksums, and that should give you audio, and most packets will go through. On Uniden scanners, and on DSDPlus, they ignore CRC for these cases automatically.

kazzaw, 2022-11-15:
Yes ignore CRC Checksums is enabled

Nokoa, 2022-11-15:
Could it be simplex transmissions (MS)?

kazzaw, 2022-11-15:
DSD is reporting as BS

Nokoa, 2022-11-15:
That's really weird. Could you capture baseband recordings of this so I can try?

DSheirer, 2023-02-11:
Issue resolved under #1422 with PR: https://github.com/DSheirer/sdrtrunk/pull/1442

There was an issue where the CRC checksum was not being calculated correctly for link control messages and the MotoTRBO TXI bit and Cap+ bit was not being parsed correctly.

closed 2023-02-11 by DSheirer