DSheirer/sdrtrunk #1271: Feature Request: P25 Packet View

catchxray, 2022-06-06:
Hello and thank you for all of your hard work on this project. I was going to ask for a certain type of encryption support however after looking through other posts it is clear that you have no desire to add P25 decryption support and that is respected. With that being the case, would it be possible to add a feature to decode and view actual packet frames within P25 so that further packet sniffing on other software is easier to complete? 
Examples: 
Header- Raw Data
LDU1- Raw Data
LDU2- Raw Data
etc..
Thank you for your time.

Nokoa, 2022-06-08:
I believe that SDRTrunk is able to save a file with the IMBE voice frames. Those voice frames are in a different form, but are all there, you have to do some stuff to them, bit shifting, etc. But all the information is there. Including the error correction bits, which after application are not needed to apply your keystream. What algorithm are you trying to implement? ADP, DES, AES?

catchxray, 2022-06-11:
I am attempting RC4/ADP decryption support. Not to break the keys but to be able to view key data in the key stream. This is an academic research project. I have seen in other posts using another form of SDR decoder that it is possible that keys are in plain view. I would like to test this with this software.