DSheirer/sdrtrunk #1272: Audio turns to garbage mid-transmission

doug-hoffman, 2022-06-08:
Experiencing intermittent issues with audio decoding on a P25 phase 2 system. The issue occurs at the exact same time on two different computers with two different SDRs, two different antennas, in two locations (about 50ft apart).

The problem has been ongoing for a while and there is usually no indication that system users are having problems receiving.

Here's an example: https://calls.broadcastify.com/1085/1654713523-20055.mp3

I also have the tuner baseband recording that includes this instance available from both computers.

davidbelle, 2022-06-09:
I have seen this with computers that don't have enough memory or the CPU is working too hard and can't keep up. Have a look at your system resources when that happen (if it happens often enough to be able to catch it).

doug-hoffman, 2022-06-09:
My streaming box is an i7-8665UE (4x1.7GHz) with 32GB RAM, running with ```-Xmx24G -Xms24G```. Load average is ~1. SDRTrunk usually falls somewhere in 100-150% CPU usage. Most of the time, the CPU is even running at a lower clock speed:

```
$ cat /proc/cpuinfo | grep MHz
cpu MHz		: 1000.040
cpu MHz		: 1000.014
cpu MHz		: 1000.144
cpu MHz		: 1000.040
cpu MHz		: 999.997
cpu MHz		: 1000.004
cpu MHz		: 1000.056
cpu MHz		: 1000.000
```

Garbage collection is only occurring about every 15 seconds, so no obvious memory issues:

```
[2022-06-09T09:37:24.970-0400][90262.643s][1796839] GC(5763) Pause Young (Normal) (G1 Evacuation Pause) 15043M->355M(24576M) 11.299ms
[2022-06-09T09:37:39.166-0400][90276.839s][1796839] GC(5764) Pause Young (Normal) (G1 Evacuation Pause) 15043M->355M(24576M) 8.177ms
[2022-06-09T09:37:53.310-0400][90290.983s][1796839] GC(5765) Pause Young (Normal) (G1 Evacuation Pause) 15043M->355M(24576M) 7.779ms
[2022-06-09T09:38:08.943-0400][90306.616s][1796839] GC(5766) Pause Young (Normal) (G1 Evacuation Pause) 15043M->355M(24576M) 7.432ms
```

My dev machine that I was testing on is a Mac Mini i7 6x3.2GHz with 32GB RAM. That should rule out any CPU concerns.

The fact that it was occurring simultaneously (or as simultaneously as my brain could perceive while listening to both in realtime) on two different machines means it's almost certainly either an RF issue or a fault in SDRTrunk. I would expect an RF issue to be noticed by actual users of the system.

davidbelle, 2022-06-09:
Yep I have to agree with you then. Not a resource issue.

Ctpolicelive, 2022-06-10:
I seem to get this same digital distortion when the weather heats up. Once the room temp gets over 80degrees my RTL-SDR's don't like it at all and I will get this same digital distortion randomly. Sometimes, when its really hot outside, even while the PC is in the cool ac, it will still happen. Very hot summer days produce this effect for me.

doug-hoffman, 2022-06-10:
Interesting. My streaming box is located in my basement, which averages 65°F, and my RTL-SDRs were sandwiched between 4x6in heatsinks, which kept them cold to the touch.

I recently switched to Airspy R2s with no change. I've also experienced the problem with an Airspy mini.

Ctpolicelive, 2022-06-10:
It's a odd issue. I seem to deal with it when the leaves grow in, in the spring, and during the summer heat.

doug-hoffman, 2022-06-10:
Sounds like your issues are more RF related. Either a decrease in sensitivity as the temperature rises or decrease in received signal strength due to obstructions.

In my case, I can see the tower at one of the transmitters across the valley (it's a simulcast site). Not a thing in the way. Signal strength is consistently within a few dB on each frequency.

Ctpolicelive, 2022-06-10:
I think so.. I think it's a sensitivity thing in my case. But I can also see one of my simulcast towers out my window as well. So it really beats me what causes this issue. Other then heat and the tree leaves. I know for sure we aren't the only ones who deal with this issue. CALLS is also aware of this issue when they apply duplicates into the system. Who knows if there is a fix.

doug-hoffman, 2022-06-10:
For reference, this is what the spectrum looked like around the time I was having problems:

[image]

For some reason I can't get anything to decode when using a recording tuner with the tuner baseband, so I've manually defined the first 3 voice channels to indicate what is/isn't part of the system I'm monitoring. One of the neighboring counties has a P25 system on adjacent frequencies, but I've not yet seen a reason to blame the problem on that.

doug-hoffman, 2022-06-10:
Adding some debug to find out where things are going wrong...

edac/BerlekempMassey.java:
```
                {
                    /* no. roots != degree of elp => >tt errors and cannot solve */
                    mLog.error("no. roots != degree of elp => >tt errors and cannot solve");
                    irrecoverable_error = true;
                }
```

yields:

```
12:33:41.110 ERROR i.g.d.e.BerlekempMassey - no. roots != degree of elp => >tt errors and cannot solve  [186MB/316MB 59%]
12:33:41.361 ERROR i.g.d.e.BerlekempMassey - no. roots != degree of elp => >tt errors and cannot solve  [190MB/316MB 60%]
12:33:41.362 ERROR i.g.d.e.BerlekempMassey - no. roots != degree of elp => >tt errors and cannot solve  [190MB/316MB 60%]
12:33:41.363 ERROR i.g.d.e.BerlekempMassey - no. roots != degree of elp => >tt errors and cannot solve  [190MB/316MB 60%]
12:33:41.463 ERROR i.g.d.e.BerlekempMassey - no. roots != degree of elp => >tt errors and cannot solve  [192MB/316MB 60%]
12:33:41.710 ERROR i.g.d.e.BerlekempMassey - no. roots != degree of elp => >tt errors and cannot solve  [195MB/316MB 61%]
12:33:41.711 ERROR i.g.d.e.BerlekempMassey - no. roots != degree of elp => >tt errors and cannot solve  [195MB/316MB 61%]
12:33:41.711 ERROR i.g.d.e.BerlekempMassey - no. roots != degree of elp => >tt errors and cannot solve  [195MB/316MB 61%]
12:33:41.813 ERROR i.g.d.e.BerlekempMassey - no. roots != degree of elp => >tt errors and cannot solve  [197MB/316MB 62%]
12:33:42.060 ERROR i.g.d.e.BerlekempMassey - no. roots != degree of elp => >tt errors and cannot solve  [201MB/316MB 63%]
12:33:42.060 ERROR i.g.d.e.BerlekempMassey - no. roots != degree of elp => >tt errors and cannot solve  [201MB/316MB 63%]
12:33:42.061 ERROR i.g.d.e.BerlekempMassey - no. roots != degree of elp => >tt errors and cannot solve  [201MB/316MB 63%]
12:33:42.162 ERROR i.g.d.e.BerlekempMassey - no. roots != degree of elp => >tt errors and cannot solve  [202MB/316MB 64%]
12:33:42.411 ERROR i.g.d.e.BerlekempMassey - no. roots != degree of elp => >tt errors and cannot solve  [205MB/316MB 65%]
```

Which continues as long as the loop repeats. It appears to get stuck in a state where it thinks it has to give up, even though subsequent data is valid.

DSheirer, 2026-11-08:
This is likely resolved with recent updates.

closed 2026-11-08 by DSheirer