DSheirer/sdrtrunk #1273: SDR Trunk DMR Overlap Issue

Seagravebuff60, 2022-06-10:
Have been having a few issues with SDR Trunk Recently. It's very complicated and hard to explain but ill do it the best way I can.

I have 2 Separate Systems in SDR Trunk, Both systems are DMR Connect + systems but it seems that System A and System B are bleeding thru on Each other, and SDR Trunk is getting confused about what system is which, and mixing up the aliases and Radio IDs. 

What I mean is that a TG and RID that belong to System A are showing up tagged as System B with a wrong frequency and Vise versa, all the time. Sometimes, SDR Trunk will get so confused and Show a TG and RID That belong to System A as System B and show up at the exact same time. The 2 Systems are 100% Confirmed to the usage and are not interconnected, they are 2 separate systems. I then had one of my friends try this out on his setup. He's receiving the same as DMR Systems I am, Different sites tho for each system. And the same overlap issue is happening on his SDR Trunk setup. So it seems like I'm not the only one having this issue. 

I first noticed this issue on Trunking Recorder, But I narrowed it down to SDR Trunk feeding False info To Trunking Recorder. 

Does anyone know what might be going on? I'm running SDR Trunk v0.5.0- Alpha 6. And the latest trunking recorder beta.

doug-hoffman, 2022-06-10:
I saw similarly weird behavior in alpha 6 with a Cap+ DMR system getting mixed in with a P25 system that I was monitoring at the same time. Sometimes it would fail to even resolve aliases that existed and worked seconds before.

I defined each individual frequency as its own channel and the problem went away. Still run it that way today and it's still solid. (Haven't tried with multiple freqs sharing a channel lately, so I don't know if the problem was ever corrected.)

Nokoa, 2022-07-11:
Are the aliases correct in the saved files?