DSheirer/sdrtrunk #1280: NO AUDIO ON SCANNER FEED for 3 days

Toto411, 2022-07-05:
Hopefully this is the right place to post this. 
I've been using broadcastify app for 5 yrs, listening to the same feed. Never had an issue until 2 weeks ago when there was no audio. 
It came back on a few days ago, it's not working again today (July 4th)
Is there an alternate way to get this feed?
⬇️⬇️⬇️⬇️⬇️⬇️⬇️⬇️⬇️⬇️⬇️⬇️⬇️⬇️
Rutherford County Sheriff, Fire and EMS
Rutherford County VIPER system. Sheriff, Forest City PD, RPD, NCSHP Troop G, Fire, & EMS
Location: Rutherford County, North Carolina, United States

Broadcasting VIPER feeds for Rutherford County, North Carolina. Broadcasting from Spindale VIPER tower. Control frequency is 853.850.

Rutherford EMS 41001
Rutherford Fire 41002
Rutherford Sheriff 41003
Rutherfordton PD 41013
Forest City PD 41020
NCSHP Troop G 52198

Using SDRtrunk (https://github.com/DSheirer/sdrtrunk/) and (3) RTL-SDR R820T2/RTL2832U USB SDRs

DSheirer, 2022-07-05:
Wrong forum.  Please report this to broadcastify service provider.  It's not an sdrtrunk issue.

closed 2022-07-05 by DSheirer

Toto411, 2022-07-05:
It says you're the service provider.

Angie Hughes

DSheirer, 2022-07-05:
Angie, Broadcastify is the service provider.