DSheirer/sdrtrunk #1281: M1  issue for NBFM/analog channels

geoffb1723, 2022-07-07:
Not sure what is going on, the app works fine for P25 on M1. But all NBFM analog channels now no longer receive, it appears there is no decode, vs no audio, if I lower the squelch as far as possible, nothing, never see any signal indication. This only started after moving to a MacStudio using M1 20 core. 

Here's a startup log, despite it being an ARM release, the Host OS Arch shows as x86_64. A minor nit, it throws an exception starting the GUI but does start.  JRE 18.0.1, JSE same ver, Oracle versions. Thanks for looking at this !

WARNING: Using incubator modules: jdk.incubator.vector
WARNING: package com.sun.java.swing.plaf.windows not in java.desktop
08:26:59.509 INFO  i.g.d.icon.IconModel - loading icons file [/Users/geoffb/SDRTrunk/settings/icons.xml]  [16MB/2GB 0%]
08:27:01.200 INFO  i.g.d.log.ApplicationLog - Application Log File: /Users/geoffb/SDRTrunk/logs/sdrtrunk_app.log  [8MB/112MB 7%]
08:27:01.238 INFO  i.g.d.log.ApplicationLog - SDRTrunk Version  : 0.5.0-beta3  [9MB/112MB 8%]
08:27:01.240 INFO  i.g.d.log.ApplicationLog - Gradle Version    : Gradle 7.4.2  [10MB/112MB 9%]
08:27:01.241 INFO  i.g.d.log.ApplicationLog - Build Timestamp   : 2022-05-04T14:55:06.134-0400  [10MB/112MB 9%]
08:27:01.241 INFO  i.g.d.log.ApplicationLog - Build-JDK         : 18.0.1 (BellSoft 18.0.1+12  [10MB/112MB 9%]
08:27:01.241 INFO  i.g.d.log.ApplicationLog - Build OS          : Linux (amd64 5.13.0-40-generic  [10MB/112MB 9%]
08:27:01.241 INFO  i.g.d.log.ApplicationLog -   [10MB/112MB 9%]
08:27:01.241 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [10MB/112MB 9%]
08:27:01.241 INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application ***  [10MB/112MB 9%]
08:27:01.241 INFO  i.g.d.log.ApplicationLog - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [10MB/112MB 9%]
08:27:01.241 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [10MB/112MB 9%]
08:27:01.242 INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%]  [11MB/112MB 10%]
08:27:01.244 INFO  i.g.d.log.ApplicationLog - Host OS Name:          Mac OS X  [11MB/112MB 10%]
08:27:01.244 INFO  i.g.d.log.ApplicationLog - Host OS Arch:          x86_64  [11MB/112MB 10%]
08:27:01.244 INFO  i.g.d.log.ApplicationLog - Host OS Version:       12.4  [11MB/112MB 10%]
08:27:01.245 INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        20  [11MB/112MB 10%]
08:27:01.245 INFO  i.g.d.log.ApplicationLog - Host Max Java Memory:  29 GB  [11MB/112MB 10%]
08:27:01.245 INFO  i.g.d.log.ApplicationLog - Storage Directories:  [11MB/112MB 10%]
08:27:01.245 INFO  i.g.d.log.ApplicationLog -  Application Root: /Users/geoffb/SDRTrunk  [11MB/112MB 10%]
08:27:01.247 INFO  i.g.d.log.ApplicationLog -  Application Log:  /Users/geoffb/SDRTrunk/logs  [11MB/112MB 10%]
08:27:01.248 INFO  i.g.d.log.ApplicationLog -  Event Log:        /Users/geoffb/SDRTrunk/event_logs  [11MB/112MB 10%]
08:27:01.249 INFO  i.g.d.log.ApplicationLog -  Playlist:         /Users/geoffb/SDRTrunk/playlist  [11MB/112MB 10%]
08:27:01.250 INFO  i.g.d.log.ApplicationLog -  Recordings:       /Users/geoffb/SDRTrunk/recordings  [11MB/112MB 10%]
08:27:01.414 INFO  i.g.d.util.ThreadPool - Application thread pool created SCHEDULED and CACHED executors threads  [15MB/112MB 13%]
08:27:01.420 INFO  i.g.dsheirer.gui.SDRTrunk - Home path: /Users/geoffb/SDRTrunk  [15MB/112MB 13%]
08:27:01.424 INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [/Users/geoffb/SDRTrunk/SDRTrunk.properties]  [15MB/112MB 13%]
08:27:01.424 INFO  i.g.d.p.SystemProperties - SystemProperties - application properties loaded [/Users/geoffb/SDRTrunk/SDRTrunk.properties]  [15MB/112MB 13%]
08:27:01.548 INFO  i.g.d.s.t.m.TunerManager - Discovering tuners ...  [19MB/112MB 17%]
08:27:01.773 INFO  i.g.d.s.t.m.TunerManager - LibUsb API Version: 1.0.262  [19MB/112MB 17%]
08:27:01.776 INFO  i.g.d.s.t.m.TunerManager - LibUsb Version: 1.0.22.11312  [19MB/112MB 17%]
08:27:01.797 INFO  i.g.d.s.t.m.TunerManager - LibUsb - discovered [47] potential usb devices  [19MB/112MB 17%]
08:27:01.802 INFO  i.g.d.s.t.m.TunerManager - Discovered tuner at USB Bus [12] Port [1.2.1.3] Tuner Class [RTL-2832]  [19MB/112MB 17%]
08:27:01.803 INFO  i.g.d.s.t.m.TunerManager - Discovered tuner at USB Bus [12] Port [1.2.1.4] Tuner Class [RTL-2832]  [19MB/112MB 17%]
08:27:01.804 INFO  i.g.d.s.t.m.TunerManager - Discovered tuner at USB Bus [13] Port [3] Tuner Class [RTL-2832]  [19MB/112MB 17%]
08:27:01.806 INFO  i.g.d.s.t.m.TunerManager - Tuner: USB Tuner - RTL-2832 USB Bus:12 Port:1.2.1.3 - Added / Starting ...  [21MB/112MB 18%]
08:27:03.026 INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [26MB/112MB 23%]
08:27:03.029 INFO  i.g.d.s.t.m.TunerManager - Tuner: USB Tuner - RTL-2832 USB Bus:12 Port:1.2.1.3 - Applying Tuner Configuration  [26MB/112MB 23%]
08:27:03.293 INFO  i.g.d.s.t.m.TunerManager - Tuner: USB Tuner - RTL-2832 USB Bus:12 Port:1.2.1.4 - Added / Starting ...  [27MB/112MB 24%]
08:27:04.484 INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [27MB/112MB 24%]
08:27:04.488 INFO  i.g.d.s.t.m.TunerManager - Tuner: USB Tuner - RTL-2832 USB Bus:12 Port:1.2.1.4 - Applying Tuner Configuration  [27MB/112MB 24%]
08:27:04.722 INFO  i.g.d.s.t.m.TunerManager - Tuner: USB Tuner - RTL-2832 USB Bus:13 Port:3 - Added / Starting ...  [27MB/112MB 24%]
08:27:05.874 INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [28MB/112MB 25%]
08:27:05.879 INFO  i.g.d.s.t.m.TunerManager - Tuner: USB Tuner - RTL-2832 USB Bus:13 Port:3 - Applying Tuner Configuration  [28MB/112MB 25%]
08:27:06.127 INFO  i.g.d.s.t.m.TunerManager - LibUsb - starting HotPlug detection  [28MB/112MB 25%]
08:27:06.155 INFO  i.g.d.s.t.m.TunerManager - Tuner plug-in detected at USB Bus [12] Port [3] Tuner Class [RTL-2832]  [28MB/112MB 25%]
08:27:06.156 INFO  i.g.d.s.t.m.TunerManager - Tuner plug-in detected at USB Bus [12] Port [4] Tuner Class [RTL-2832]  [28MB/112MB 25%]
08:27:06.157 INFO  i.g.d.s.t.m.TunerManager - Tuner plug-in detected at USB Bus [13] Port [3] Tuner Class [RTL-2832]  [28MB/112MB 25%]
08:27:06.162 INFO  i.g.d.s.SettingsManager - SettingsManager - loading settings file [/Users/geoffb/SDRTrunk/settings/settings.xml]  [28MB/112MB 25%]
08:27:06.640 WARN  i.g.d.v.VectorUtilities - CPU supports maximum SIMD instructions of Species[float, 4, S_128_BIT]  [41MB/112MB 37%]
08:27:08.569 INFO  i.g.d.p.PlaylistManager - Loading playlist [/Users/geoffb/SDRTrunk/playlist/gb.xml]  [24MB/112MB 21%]
08:27:09.099 INFO  i.g.dsheirer.gui.SDRTrunk - starting main application gui  [36MB/112MB 32%]
08:27:42.951 ERROR i.g.d.g.p.r.SystemEditor - Error retrieving system information  [262MB/384MB 68%]
java.lang.NullPointerException: Cannot read field "left" because "<local2>" is null
	at java.base/java.util.TreeMap.rotateLeft(Unknown Source)
	at java.base/java.util.TreeMap.fixAfterInsertion(Unknown Source)
	at java.base/java.util.TreeMap.addEntry(Unknown Source)
	at java.base/java.util.TreeMap.put(Unknown Source)
	at java.base/java.util.TreeMap.put(Unknown Source)
	at io.github.dsheirer.rrapi.RadioReferenceService.getVoicesMap(RadioReferenceService.java:894)
	at io.github.dsheirer.gui.playlist.radioreference.SystemEditor.initRadioReferenceDecoder(SystemEditor.java:339)
	at io.github.dsheirer.gui.playlist.radioreference.SystemEditor.lambda$setSystem$5(SystemEditor.java:284)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.base/java.lang.Thread.run(Unknown Source)
08:27:56.887 INFO  i.g.d.a.c.m.JmbeAudioModule - Loading JMBE library from [/Users/geoffb/jmbe/creator-osx-aarch64-v1.0.9/jmbe-1.0.9.jar]  [265MB/384MB 69%]
08:27:56.922 INFO  i.g.d.a.c.m.JmbeAudioModule - JMBE audio conversion library loaded: JMBE Audio Conversion Library v1.0.9  [265MB/384MB 69%]
08:27:56.922 INFO  i.g.d.a.c.m.ImbeAudioModule - JMBE audio conversion library IMBE CODEC successfully loaded - P25-1 audio will be available  [265MB/384MB 69%]
08:27:56.945 INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [266MB/384MB 69%]
08:28:17.185 INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [766MB/1GB 63%]
08:28:36.831 INFO  i.g.d.s.t.m.FrequencyErrorCorrectionManager - Auto-Correcting Tuner PPM to [-2.0]  [701MB/1GB 57%]
08:28:57.365 INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [734MB/1GB 60%]

Toto411, 2022-07-07:
> It if helps, redid the SIMD Cal and am using polyphase chanelizer.

It's back on!! I guess it helped!
THANK YOU SO MUCH!!

Angie Hughes

geoffb1723, 2022-07-07:
Did you mean the NBFM too ? ALso, you are welcome, I didn't really do very much !

DSheirer, 2022-07-08:
@geoffb1723 Are you using the ARM (M1) or x86-64 release of sdrtrunk on your M1?

This seems to indicate that you're using the x86-64 version:
08:27:01.244 INFO i.g.d.log.ApplicationLog - Host OS Arch: x86_64 [11MB/112MB 10%]

geoffb1723, 2022-07-08:
ok, remove foot from mouth, when I migrated over to the new machine, the migration moved the old x86 version over on top of the ARM version I had downloaded. Let me redownload the ARM version and see what gives. I  will follow up shortly, got a few meetings to do this am.
Thanks for a really great program, really excellent.

geoffb1723, 2022-07-08:
Remove Size 11  from mouth. MIgration from old machine overwrote a clean install of ARM version. Let me get on to the ARM release and I will get back to you how things are working.
Thanks for a great piece of work, really appreciated.

geoffb1723, 2022-07-08:
Interesting, when I try to start the ARM native, I get an exception. There is nothing like a CLASSPATH  override in my .zshrc. Going back to the x86 ver, it runs.. Hmmm.. macos 12.4.
Appreciate your time in looking at this - g

WARNING: Using incubator modules: jdk.incubator.vector
WARNING: package com.sun.java.swing.plaf.windows not in java.desktop
06:40:32.956 INFO  i.g.d.icon.IconModel - loading icons file [/Users/xx/SDRTrunk/settings/icons.xml]  [16MB/2GB 0%]
06:40:33.277 INFO  i.g.d.log.ApplicationLog - Application Log File: /Users/xx/SDRTrunk/logs/sdrtrunk_app.log  [8MB/112MB 7%]
06:40:33.286 INFO  i.g.d.log.ApplicationLog - SDRTrunk Version  : 0.5.0-beta3  [9MB/112MB 8%]
06:40:33.287 INFO  i.g.d.log.ApplicationLog - Gradle Version    : Gradle 7.4.2  [9MB/112MB 8%]
06:40:33.287 INFO  i.g.d.log.ApplicationLog - Build Timestamp   : 2022-05-04T14:55:06.134-0400  [9MB/112MB 8%]
06:40:33.287 INFO  i.g.d.log.ApplicationLog - Build-JDK         : 18.0.1 (BellSoft 18.0.1+12  [9MB/112MB 8%]
06:40:33.287 INFO  i.g.d.log.ApplicationLog - Build OS          : Linux (amd64 5.13.0-40-generic  [9MB/112MB 8%]
06:40:33.287 INFO  i.g.d.log.ApplicationLog -   [9MB/112MB 8%]
06:40:33.287 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [9MB/112MB 8%]
06:40:33.287 INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application ***  [9MB/112MB 8%]
06:40:33.287 INFO  i.g.d.log.ApplicationLog - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [9MB/112MB 8%]
06:40:33.287 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [11MB/112MB 10%]
06:40:33.288 INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%]  [11MB/112MB 10%]
06:40:33.288 INFO  i.g.d.log.ApplicationLog - Host OS Name:          Mac OS X  [11MB/112MB 10%]
06:40:33.288 INFO  i.g.d.log.ApplicationLog - Host OS Arch:          aarch64  [11MB/112MB 10%]
06:40:33.288 INFO  i.g.d.log.ApplicationLog - Host OS Version:       12.4  [11MB/112MB 10%]
06:40:33.288 INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        20  [11MB/112MB 10%]
06:40:33.288 INFO  i.g.d.log.ApplicationLog - Host Max Java Memory:  29 GB  [11MB/112MB 10%]
06:40:33.288 INFO  i.g.d.log.ApplicationLog - Storage Directories:  [11MB/112MB 10%]
06:40:33.288 INFO  i.g.d.log.ApplicationLog -  Application Root: /Users/xx/SDRTrunk  [11MB/112MB 10%]
06:40:33.289 INFO  i.g.d.log.ApplicationLog -  Application Log:  /Users/xx/SDRTrunk/logs  [11MB/112MB 10%]
06:40:33.289 INFO  i.g.d.log.ApplicationLog -  Event Log:        /Users/xx/SDRTrunk/event_logs  [11MB/112MB 10%]
06:40:33.289 INFO  i.g.d.log.ApplicationLog -  Playlist:         /Users/xx/SDRTrunk/playlist  [11MB/112MB 10%]
06:40:33.289 INFO  i.g.d.log.ApplicationLog -  Recordings:       /Users/xx/SDRTrunk/recordings  [11MB/112MB 10%]
06:40:33.333 INFO  i.g.d.util.ThreadPool - Application thread pool created SCHEDULED and CACHED executors threads  [14MB/112MB 13%]
06:40:33.334 INFO  i.g.dsheirer.gui.SDRTrunk - Home path: /Users/x/SDRTrunk  [14MB/112MB 13%]
06:40:33.334 INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [/Users/xx/SDRTrunk/SDRTrunk.properties]  [14MB/112MB 13%]
06:40:33.334 INFO  i.g.d.p.SystemProperties - SystemProperties - application properties loaded [/Users/geoffb/SDRTrunk/SDRTrunk.properties]  [14MB/112MB 13%]
06:40:33.368 INFO  i.g.d.s.t.m.TunerManager - Discovering tuners ...  [17MB/112MB 15%]
Exception in thread "main" java.lang.UnsatisfiedLinkError: Can't load library: /var/folders/qt/8n2xgkdx1lv8kzv3n_vtmrm00000gn/T/usb4java13675722538697338641.tmp/libusb4java.dylib
	at java.base/java.lang.ClassLoader.loadLibrary(Unknown Source)
	at java.base/java.lang.Runtime.load0(Unknown Source)
	at java.base/java.lang.System.load(Unknown Source)
	at org.usb4java.Loader.load(Loader.java:323)
	at org.usb4java.LibUsb.<clinit>(LibUsb.java:690)
	at io.github.dsheirer.source.tuner.manager.TunerManager.start(TunerManager.java:99)
	at io.github.dsheirer.gui.SDRTrunk.<init>(SDRTrunk.java:177)
	at io.github.dsheirer.gui.SDRTrunk.main(SDRTrunk.java:714)

cfc62, 2022-07-25:
Try changing your decode from polyphase to heterodyne (or vice-versa) and restart SDRT; that's cleared the NBFM no decode issue for others.

geoffb1723, 2022-07-26:
Ok will do, many thanks. Stby one pls.

Hmm, I have 3 NBFM channels enabled, I changed from polyphase to heterodyne, restarted numerous times. No change.
Recreated the NBFM channels, to make sure I had a clean channel config.

If I look at each RTL ( I use 3), they are at -60 or -66 dBm, squelch wise, locked, let me mess with the sample rate, in order to update the settings.
Interesting, there’s no fluctuating level/RF signal indicated on the 'channel’ RF meter, as in no receive baseline level of note, period. Never shows anything.
Looks like they are not down converting anything much ?

Huh. Interesting, if I look at the waterfall/pan, it’s vague fuzzy noise, carriers DO weakly emerge from the noise with a red line (but very, very weak). Never comes up
to about around -80 dBm. Not sure what’s going on. But, it looks like the sensitivity is way down the curve. Never breaks any squelch or any indication
at all  on the level meter. So it IS doing something, I was wrong, just it’s gain squashed.

Appreciate your help as always ! Highly appreciated.

Wonder if this an M1issue, I dont know enough to diagnose the DSP side of things.
I am not a DSP guy, I am a network guy. The DSP side of things, I have no idea. A double float, you are on your own man !

This is us, we are not total (only partial) morons. https://dmr.wa6ycz.org/
Unlike most hams, we bang out code and dont complain.  I could care less about talking with
Boris in Transylvania, we all like to build stuff.  8 mountain tops, dozens of interesting sensors/nets. PM me for
a few links to have fun with, like our private seismic nets, NASA Deep Sky Fireball (comet) network cams, ADSB, AIS
and other receivers,  etc. We run a TON of stuff around the Bay Area.

You aware of this to watch the fires ? Like 3 500 cams   https://www.alertwildfire.org/ Go to “regions”.
Amazing.

rgds

g

closed 2026-11-06 by DSheirer