DSheirer/sdrtrunk #1289: Missing ALSA audio playback devices

Ventriduct, 2022-07-19:
I am unable to play audio with SDR Trunk. It is missing several audio devices.

Here are screenshots of the audio devices shown:
[image]
[image]

Here is the full list of audio devices from aplay -l:
[aplay-l.txt](https://github.com/DSheirer/sdrtrunk/files/9142811/aplay-l.txt)
from aplay -L:
[aplay-L.txt](https://github.com/DSheirer/sdrtrunk/files/9142810/aplay-L.txt)

SDR Trunk doesn't show any of the pipewire, pulse, or Scarlett 2i2 devices. It only shows some Nvidia devices and devices called generic.

It's receiving signals fine:
[image]

Here is the terminal output:
[SDRTrunk_Output.txt](https://github.com/DSheirer/sdrtrunk/files/9142793/SDRTrunk_Output.txt)

I'm using SDR Trunk linux x86_64 v0.4.0 on Arch Linux, kernel version 5.18.10-arch1-1.

coolacid, 2023-07-03:
Having the same issue on Ubuntu 22.10 - I suspect this is due to the move from PulseAudio to PipeWire - There is lots of articles related to other Java applications not working - however those workarounds don't work with SDR Trunk.

coolacid, 2023-07-03:
Upgraded to 23.04 and default sound device is there. I want to say I did have this working under 22.10 - I should have just tried a reboot before upgrading.