DSheirer/sdrtrunk #1296: Instability in Java after a couple hours in Beta 5

Brad-git-man, 2022-07-29:
I have noticed in Beta 5, that after a couple hours of running SDRT that java becomes unstable and starts putting 80% load on my 16 core CPU. I have to kill it off with Task Manager at that time to stop the insane CPU consumption. This didn't happen in prior versions of SDRT.

GTR8000, 2022-08-02:
Sorry to be "that guy", but I'm running beta5 on five vastly different Win 10 machines, some of them very modern and powerful, and some 12 year old relics. All decoding busy P25P2 systems and not one of them has experienced any issue like what you're describing, and they've all been running continuously since beta5 was released a week ago.

jx34tech, 2022-08-16:
I can say that I have experienced a similar issue as @Brad-git-man , but for me it more occurred after a few days, although depending on the activity of the system, although increasing the RAM given to the JVM appears to fix it and it lasts much longer before locking up and requiring a restart

closed 2026-11-06 by DSheirer