DSheirer/sdrtrunk #1297: Suggestion to save settings under View drop-down menu

Deziel0495, 2022-08-01:
Would it be possible to have the options under the "View" drop-down of "disable spectrum & waterfall" and "show streaming status" saved, so it will remember your preference when launching the program.

GTR8000, 2022-08-02:
Those two settings should already persist through restarts of the software. Has worked on mine going all the way back to early versions of 0.5.0 and continues to work right up to beta5.

Deziel0495, 2022-08-02:
Hmm. I'm using the latest, 0.5.0 Beta 5 and it's not persistent. Wasn't
with Beta 4 either. Every time I restart the software I turn off the
waterfall and turn on streaming status.

bnelson333, 2022-09-13:
Throwing my vote in on this for the waterfall persist. sdrtrunk used to remember that i turned the waterfall off. if i relaunched, it would stay off. somewhere between alpha and beta that setting no longer persists for me and it's a huge resource hog. I'm using this on xubuntu 18.04.

bnelson333, 2022-09-13:
I did some digging on mine and I think there might be a bug here. The file /home/user/SDRTrunk/SDRTrunk.properties has a line that says:

`spectral.display.enabled=false
`

But when starting up sdrtrunk it still turns on the waterfall. Permissions seem ok, it seems to be able to read and write to that file. I even chmodded it to 777 just to check and no change. Is sdrtrunk not correctly reading that line out of the config file?

GTR8000, 2022-09-13:
I'm baffled as to why you are both having issues with these settings not persisting. I'm running SDRTrunk 0.5.0 beta5 on four Win 10 computers, and both the waterfall (off) and streaming status (on) states have persisted throughout many countless restarts of the program over the past year or so.

Perhaps this is a Linux vs Windows issue?

Here is the text of my SDRTrunk.properties file

`#SDRTrunk - SDR Trunking Decoder Application Settings
#Sun Sep 11 19:45:15 EDT 2022
audio.channel.panel.color.background=-16777216
audio.channel.panel.color.label=-4144960
audio.channel.panel.color.muted=-65536
audio.channel.panel.color.value=-16711936
**main.broadcast.status.visible=true**
root.directory=SDRTrunk
settings.currentFilename=settings.xml
settings.defaultFilename=settings.xml
spectral.display.dft.size=FFT04096
**spectral.display.enabled=false**
spectral.display.frame.rate=20`

bnelson333, 2022-09-13:
I think it is, would there be different code bases for the different releases? It seems this works on windows and my buddy confirmed it works on the raspberry pi (the aarch release), but it seems to not work on the linux x86 build. I remember it working back on an alpha release and then I upgraded to a beta release and it stopped working. If i get time later tonight I'll see if I can figure out which version "broke it".

bnelson333, 2022-09-13:
I confirmed just now that it most definitely broke between betas 2 and 3. Version 0.5.0 Beta 2 functions as intended (waterfall setting persists between restarts), Version 0.5.0 Beta 3 broke that functionality.

DSheirer, 2022-11-26:
Closing - recreating as subtask under issue #1348

closed 2022-11-26 by DSheirer