DSheirer/sdrtrunk #1303: JMBE falsely detecting AMBE Tone Frames in Voice Frames that contain no tones

Nokoa, 2022-08-08:
I have observed multiple instances where it appears that JMBE detects an AMBE Tone Frame in a voice frame, and plays short tones here and there in transmissions.

GTR8000, 2022-08-10:
Any time I've seen this happen, it was due to the talk permit tone coming through the console's (dispatcher's) audio very briefly at the start of transmissions. Not really anything SDRTrunk is going to do about that, as technically it is a valid tone being detected.

Nokoa, 2022-08-12:
I don't believe it's that. This happens at any time in the transmission, not only the beginning, and from any unit. Transmissions compared through scanner/ radio, such tones/ beeps are not present. I added a debug message to the JMBE AMBE tone frame detection and processing, and I see dozens of instances where that clause is true. I will try to see if I can get some examples, and their corresponding frames.

I believe this is an issue I also observed with OP25, in similar manner, at least, maybe not as frequent however.

Nokoa, 2022-09-24:
Was able to confirm this by sending the voice frames from sdrtrunk to a hardware vocoder, and compared the audio, false tones are not present through the hardware AMBE vocoder, but are through JMBE.  #822

DSheirer, 2026-11-08:
This may be resolved with latest updates to sdrtrunk and jmbe library.  Closing.

closed 2026-11-08 by DSheirer