DSheirer/sdrtrunk #1304: Add TDMA CC decoding

KF5HKI, 2022-08-11:
I don't think SDRtrunk has TDMA Control Channel decoding capability. I know that it works with P25 P2 with FDMA/TDMA voice channels but I can not get it to work on a System that has TDMA Control Channel.

Here is some info that may help.
https://forums.radioreference.com/threads/p25-tdma-control-channel-decoding-requesting-help-from-experts.429312/
For the AEP System https://www.radioreference.com/db/sid/9847 in East Texas.

If you need more info please let me know and I will try to help in any way I can. Thanks

KF5HKI, 2022-08-27:
Update:
I was able to get it to work on a TDMA Control Channel by changing the playlist xml file.
From "<decode_configuration type="decodeConfigP25Phase1" ..." To "<decode_configuration type="decodeConfigP25Phase2" ..."

KF5HKI, 2022-08-29:
If possible add a way to make that change in the GUI.
As in being able to set the P25 mode from Phase 1 CC to Phase 2 (TDMA) CC without having to do it in the XML file.
[Screen Capture](https://drive.google.com/file/d/1H-xtag2JA1eIAKL0gg5zZb8mHPLWwqcu/view?usp=sharing)
[Link to AEP site#82](https://www.radioreference.com/db/site/40330)

Thanks..

lwvmobile, 2023-10-20:
@DSheirer 

Here is a link to a raw baseband recording from SDRTrunk of Duke Energy (Harris) P25p2 only which uses MAC_SIGNAL.

https://ufile.io/rxvsk7ka

The skinny of what the manual says is that MAC_SIGNAL (LCCH) is always unscrambled, is on DUID 13/Opcode 0, and should be treated like a SACCH frame during RS decoding, but has a CRC16 (same as TSBK) instead of a CRC12. It carries its own NET_STS_BCST and many other Control related PDUs, but they are generally all the same format as the Phase 1 OSP TSBK/MBT for abbreviated and extended. 

The manual also shows a scrambled variant of LCCH on DUID 4, but I've never seen that one before.

Just let me know if you want or need more information.

lwvmobile, 2023-10-20:
Also, forgot, in case you need it, this is WACN [91F14] SYS [2D7] NAC [01A]

DSheirer, 2024-05-04:
Closing as implemented.

closed 2024-05-04 by DSheirer