DSheirer/sdrtrunk #1316: Feature Request: Add SoapySDR support to SDRTrunk

gvimlag, 2022-09-19:
I use Raspberry-Pi "Soapy servers" to connect to my outside SDR-dongles/antennas.
Please add support for SoapySDR to SDRTrunk.

lewis-jeffery, 2022-09-19:
+1

Remote receivers would be a game changer for me…

jackcrombie, 2022-09-28:
+1
Yes please!

ivkraft, 2022-09-29:
+1 I would like to see LimeSDR

ImDroided, 2023-03-20:
Remote sdr's would be awesome. Soapy for sure

kernworks, 2025-01-27:
+1 Agreed.

Remote SDRs would be a huge game changer.

Soapy SDR is the standard at this point.

DSheirer, 2026-11-08:
I took another solution path.  I'm not planning to implement the SoapySDR interface.

nathan-signalhound, 2026-11-09:
@DSheirer I do have this work done on my fork. I just haven't had the time to make a request to pull it into yours. If you don't mind me asking what is your approach that no longer make this worth while?

DSheirer, 2026-11-09:
That library was somewhat generic. So I created Java FFM wrappers for each of the tuner libraries with recompile native library for each environment and wired in the options exposed by each library.  

I did create a test jSoapy library in my github, but didn't integrate it.

If it would help, I can probably generate the signal hound interface and sdrtrunk integration from your C headers and let you test it with actual hardware.  I turned the crank on that pattern 7x times this past weekend.

jackcrombie, 2026-11-09:
Hi @DSheirer , the reason I was interested in SoapySDR was the ability to use online receivers in SDRTrunk or remotely located receivers. SoapySDR is a well established protocol widely run and available.

Could you explain why you chose another path and does it have the same functionality?
Will it give users access of SDRTrunk access to the receivers available online or to remotely host their own receivers?

Just trying to understand your decision and if it provides the features people were looking for.

DSheirer, 2026-11-10:
I'm reopening the issue.  I hadn't considered the remote access to a network tuner option and reopening also leaves open the option for @nathan-signalhound to use this approach for adding signal hound support.

Consider using the jSoapySDR multi-platform, multi-arch library in my GitHub.  This is a similar approach I used for the other tuner support (jRTL, jAirspy, jHydraSDR, etc.)  I can update this to produce a version of the library that tracks the SoapySDR library version, if that's helpful.

nathan-signalhound, 2026-11-10:
@DSheirer sounds good to me. Thanks for opening this back up. I'll update my fork to use your jSoapySDR repo and try to create the MR a when I have a little bit of down time sometime soonish. Yeah a version that tracks the SoapySDR library version would be helpful.

nathan-signalhound, 2026-11-10:
@DSheirer Were you going to get a release build going for jSoapySDR so I can have those pulled in the same way you pull in the JHydraSDR?

DSheirer, 2026-11-12:
@nathan-signalhound I updated jSoapySDR and tagged a release, but that may not have what you're looking for.  I'm not at all familiar with SoapySDR ... the little that I've read is that you bundle in specific support ... Signal Hound in this instance.  The jSoapySDR only bakes in the usual collection of hobbiest tuners.

Something to consider ... I created a new jSignalHound repository that includes Java FFM bindings for the 3x driver libraries and I created a candidate integration in my sdrtrunk repo here: https://github.com/DSheirer/sdrtrunk/tree/signal-hound

This is all claude generated, so there may be some nuances to clear up.

I don't have a device to test against.

nathan-signalhound, 2026-11-13:
@DSheirer I can test against your candidate integration and provide feedback/fixes as I come across them. 

As for the jSoapySDR soapy provides a general interface and you basically write your own soapy drivers and then register with the soapy interface so they can by found by anyone interacting with soapy. If that makes sense. So I'll investigate how how made a general interface device specific and get back to you on that one.