DSheirer/sdrtrunk #1320: RTL-2832 with r828d - error: Unsupported/Unrecognized Tuner Type

PavelAravin, 2022-10-04:
###Hello. usb rtl-sdr RTL-2832 with r828d not working.

WARNING: Using incubator modules: jdk.incubator.vector
19:46:23.502 INFO  i.g.d.icon.IconModel - loading icons file [C:\Users\pavel\SDRTrunk\settings\icons.xml]  [11MB/256MB 4%]
19:46:23.863 INFO  i.g.d.log.ApplicationLog - Application Log File: C:\Users\pavel\SDRTrunk\logs\sdrtrunk_app.log  [16MB/256MB 6%]
19:46:23.897 INFO  i.g.d.log.ApplicationLog - SDRTrunk Version  : 0.5.0-beta5  [7MB/28MB 27%]
19:46:23.898 INFO  i.g.d.log.ApplicationLog - Gradle Version    : Gradle 7.4.2  [7MB/28MB 27%]
19:46:23.898 INFO  i.g.d.log.ApplicationLog - Build Timestamp   : 2022-07-23T08:15:59.564-0400  [7MB/28MB 27%]
19:46:23.898 INFO  i.g.d.log.ApplicationLog - Build-JDK         : 18.0.1 (BellSoft 18.0.1+12  [7MB/28MB 27%]
19:46:23.898 INFO  i.g.d.log.ApplicationLog - Build OS          : Linux (amd64 5.15.0-41-generic  [7MB/28MB 27%]
19:46:23.898 INFO  i.g.d.log.ApplicationLog -   [7MB/28MB 27%]
19:46:23.898 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [7MB/28MB 27%]
19:46:23.899 INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application ***  [7MB/28MB 27%]
19:46:23.899 INFO  i.g.d.log.ApplicationLog - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [7MB/28MB 27%]
19:46:23.899 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [7MB/28MB 27%]
19:46:23.899 INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%]  [7MB/28MB 27%]
19:46:23.899 INFO  i.g.d.log.ApplicationLog - Host OS Name:          Windows 11  [7MB/28MB 27%]
19:46:23.899 INFO  i.g.d.log.ApplicationLog - Host OS Arch:          amd64  [7MB/28MB 27%]
19:46:23.899 INFO  i.g.d.log.ApplicationLog - Host OS Version:       10.0  [7MB/28MB 27%]
19:46:23.900 INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        12  [7MB/28MB 27%]
19:46:23.900 INFO  i.g.d.log.ApplicationLog - Host Max Java Memory:  3 GB  [7MB/28MB 27%]
19:46:23.900 INFO  i.g.d.log.ApplicationLog - Storage Directories:  [7MB/28MB 27%]
19:46:23.900 INFO  i.g.d.log.ApplicationLog -  Application Root: C:\Users\pavel\SDRTrunk  [7MB/28MB 27%]
19:46:23.900 INFO  i.g.d.log.ApplicationLog -  Application Log:  C:\Users\pavel\SDRTrunk\logs  [7MB/28MB 27%]
19:46:23.901 INFO  i.g.d.log.ApplicationLog -  Event Log:        C:\Users\pavel\SDRTrunk\event_logs  [7MB/28MB 27%]
19:46:23.901 INFO  i.g.d.log.ApplicationLog -  Playlist:         C:\Users\pavel\SDRTrunk\playlist  [7MB/28MB 27%]
19:46:23.902 INFO  i.g.d.log.ApplicationLog -  Recordings:       C:\Users\pavel\SDRTrunk\recordings  [8MB/28MB 30%]
19:46:23.902 INFO  i.g.d.util.ThreadPool - Application thread pool created SCHEDULED and CACHED executors threads  [8MB/28MB 30%]
19:46:23.903 INFO  i.g.dsheirer.gui.SDRTrunk - Home path: C:\Users\pavel\SDRTrunk  [8MB/28MB 30%]
19:46:23.908 INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [C:\Users\pavel\SDRTrunk\SDRTrunk.properties]  [8MB/28MB 30%]
19:46:23.909 INFO  i.g.d.p.SystemProperties - SystemProperties - application properties loaded [C:\Users\pavel\SDRTrunk\SDRTrunk.properties]  [8MB/28MB 30%]
19:46:23.944 INFO  i.g.d.s.t.m.TunerManager - Discovering tuners ...  [10MB/28MB 37%]
19:46:24.039 INFO  i.g.d.s.t.m.TunerManager - LibUsb API Version: 1.0.262  [11MB/28MB 42%]
19:46:24.040 INFO  i.g.d.s.t.m.TunerManager - LibUsb Version: 1.0.22.11312  [11MB/28MB 42%]
19:46:24.053 INFO  i.g.d.s.t.m.TunerManager - LibUsb - discovered [8] potential usb devices  [11MB/28MB 42%]
19:46:24.057 INFO  i.g.d.s.t.m.TunerManager - Discovered tuner at USB Bus [1] Port [11] Tuner Class [RTL-2832]  [11MB/28MB 42%]
19:46:24.059 INFO  i.g.d.s.t.m.TunerManager - Tuner: USB Tuner - RTL-2832 USB Bus:1 Port:11 - Added / Starting ...  [12MB/28MB 44%]
19:46:25.527 INFO  i.g.d.s.t.m.DiscoveredTuner - Tuner Error - Stopping - RTL-2832 USB Bus:1 Port:11 Error: Unsupported/Unrecognized Tuner Type  [15MB/28MB 54%]
19:46:25.527 INFO  i.g.d.s.t.m.DiscoveredTuner - Stopping Tuner: RTL-2832 USB Bus:1 Port:11  [15MB/28MB 54%]
19:46:25.528 ERROR i.g.d.s.t.m.DiscoveredUSBTuner - Unable to start tuner [RTL-2832] - error: Unsupported/Unrecognized Tuner Type  [15MB/28MB 54%]
19:46:25.530 INFO  i.g.d.s.t.m.TunerManager - LibUsb Hotplug event notification Is Not Supported on this platform.  [15MB/28MB 54%]
19:46:25.534 INFO  i.g.d.s.SettingsManager - SettingsManager - loading settings file [C:\Users\pavel\SDRTrunk\settings\settings.xml]  [15MB/28MB 54%]
19:46:25.735 WARN  i.g.d.v.VectorUtilities - CPU supports maximum SIMD instructions of Species[float, 8, S_256_BIT]  [8MB/32MB 27%]
19:46:26.160 INFO  i.g.d.p.PlaylistManager - Loading playlist [C:\Users\pavel\SDRTrunk\playlist\default.xml]  [18MB/32MB 56%]
19:46:26.187 INFO  i.g.dsheirer.gui.SDRTrunk - starting main application gui  [20MB/32MB 63%]

PavelAravin, 2022-10-12:
Yes, I use one of the converted TV tuners. But is this the reason for the software not to work?

dechilders, 2022-10-14:
> Yes, I use one of the converted TV tuners. But is this the reason for the software not to work?

The root cause in the other issue I linked seemed to be from the device being a cheap knock-off. Are you sure the one you purchased is a legitimate one (Does it say "RTL . SDR" on it or "RTL-SDR")? Do you have another to swap it out with to test?

dechilders, 2022-10-16:
It appears the R828D is not supported. #1315

PavelAravin, 2022-10-17:
Really. It remains to be hoped that the author will find a way to implement support for the receiver. Thank you.

ElectronicallyE, 2022-11-18:
I believe I am having the same issue. Please see the linked image:
https://imgur.com/a/aLPwljH

I think my RTL-2832u is a clone, as it features the yellow circuit board as listed under _Fake RTL-SDR Blog V3 Clones_:
https://www.rtl-sdr.com/wp-content/uploads/2021/05/clone_identification.png

My SDR works fine for other software including DSD+ (with P25 decoding), SDRSharp (SDR#) and CubicSDR. I understand that it is a clone and therefore not as accurate or reliable, but what is preventing SDRTrunk from working?

DSheirer, 2022-11-26:
@PavelAravin R828D tuner type is not supported by sdrtrunk.

I don't have access to an sdr dongle with that type of tuner, so it would be difficult for me to add support.

closed 2022-11-26 by DSheirer

megapro17, 2023-05-30:
What the problem with adding support, if it is identical, without any improvements compared to a R820T one? https://www.rtl-sdr.com/new-rtl-sdr-tuner-chip-r828d/

megapro17, 2023-06-01:
[image]
It works?

bradjtrammell, 2023-07-30:
@DSheirer I just realized both of us are located in Central NY. I have a plethora of these adapters that I generally use to setup ADS-B receivers for people. If I can loan you one or two to help speed up integration or addition of support let me know.

These are becoming more and more prevalent and it's a shame there is no support for these.

davidmckenzie, 2023-08-18:
Just dropping a note that the new RTL-SDR.com blog dongle (one of the most popular dongles) has moved to r828d in their latest release:

https://www.rtl-sdr.com/rtl-sdr-blog-v4-dongle-initial-release/

megapro17, 2023-08-18:
Lmao they ran out of these ancient chips

Cpw302, 2023-09-02:
Megapro17.  Did you get the v4 (828d) to work on sdrtrunk?  If so, I’d be interested in how you were able to get it up and running.  I’ve had no issue using the 828d with SDR++ or sharp, but can’t get it going on SDRTrunk. Thx