DSheirer/sdrtrunk #1321: RTL2832U R820T2 Unrecognized

ecrist, 2022-10-05:
I purchased some new RTL SDR devices and they have an R820T2 instead of the R820T, and sdrtrunk lists them as unrecognized on my Windows computer.  I tried using them earlier today on my Mac, but they failed to sync (and I had to set PPM to 49.0 to align frequency).  

The markings on the SDR specify:
``` 
RTL.SDR
DVB-T+DAB+FM+SDR
RTL2832U R820T2 TCXO+BIAS T+HF
```

Here are the logs on the Windows side:

```
WARNING: Using incubator modules: jdk.incubator.vector
14:55:29.821 INFO  i.g.d.icon.IconModel - loading icons file [C:\Users\ecrist\SDRTrunk\settings\icons.xml]  [11MB/512MB 2%]
14:55:29.826 INFO  i.g.d.icon.IconModel - Icons file not found at [C:\Users\ecrist\SDRTrunk\settings\icons.xml]  [11MB/512MB 2%]
14:55:30.165 INFO  i.g.d.log.ApplicationLog - Application Log File: C:\Users\ecrist\SDRTrunk\logs\sdrtrunk_app.log  [18MB/512MB 3%]
14:55:30.191 INFO  i.g.d.log.ApplicationLog - SDRTrunk Version  : 0.5.0-beta5  [21MB/512MB 4%]
14:55:30.192 INFO  i.g.d.log.ApplicationLog - Gradle Version    : Gradle 7.4.2  [21MB/512MB 4%]
14:55:30.192 INFO  i.g.d.log.ApplicationLog - Build Timestamp   : 2022-07-23T08:15:59.564-0400  [21MB/512MB 4%]
14:55:30.192 INFO  i.g.d.log.ApplicationLog - Build-JDK         : 18.0.1 (BellSoft 18.0.1+12  [21MB/512MB 4%]
14:55:30.193 INFO  i.g.d.log.ApplicationLog - Build OS          : Linux (amd64 5.15.0-41-generic  [21MB/512MB 4%]
14:55:30.193 INFO  i.g.d.log.ApplicationLog -   [21MB/512MB 4%]
14:55:30.193 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [21MB/512MB 4%]
14:55:30.193 INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application ***  [21MB/512MB 4%]
14:55:30.193 INFO  i.g.d.log.ApplicationLog - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [21MB/512MB 4%]
14:55:30.194 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [21MB/512MB 4%]
14:55:30.194 INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%]  [21MB/512MB 4%]
14:55:30.194 INFO  i.g.d.log.ApplicationLog - Host OS Name:          Windows 10  [21MB/512MB 4%]
14:55:30.194 INFO  i.g.d.log.ApplicationLog - Host OS Arch:          amd64  [21MB/512MB 4%]
14:55:30.194 INFO  i.g.d.log.ApplicationLog - Host OS Version:       10.0  [21MB/512MB 4%]
14:55:30.195 INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        4  [21MB/512MB 4%]
14:55:30.195 INFO  i.g.d.log.ApplicationLog - Host Max Java Memory:  7 GB  [21MB/512MB 4%]
14:55:30.195 INFO  i.g.d.log.ApplicationLog - Storage Directories:  [21MB/512MB 4%]
14:55:30.195 INFO  i.g.d.log.ApplicationLog -  Application Root: C:\Users\ecrist\SDRTrunk  [21MB/512MB 4%]
14:55:30.195 INFO  i.g.d.log.ApplicationLog -  Application Log:  C:\Users\ecrist\SDRTrunk\logs  [21MB/512MB 4%]
14:55:30.196 INFO  i.g.d.log.ApplicationLog -  Event Log:        C:\Users\ecrist\SDRTrunk\event_logs  [21MB/512MB 4%]
14:55:30.197 INFO  i.g.d.log.ApplicationLog -  Playlist:         C:\Users\ecrist\SDRTrunk\playlist  [21MB/512MB 4%]
14:55:30.198 INFO  i.g.d.log.ApplicationLog -  Recordings:       C:\Users\ecrist\SDRTrunk\recordings  [21MB/512MB 4%]
14:55:30.198 INFO  i.g.d.util.ThreadPool - Application thread pool created SCHEDULED and CACHED executors threads  [21MB/512MB 4%]
14:55:30.198 INFO  i.g.dsheirer.gui.SDRTrunk - Home path: C:\Users\ecrist\SDRTrunk  [21MB/512MB 4%]
14:55:30.203 INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [C:\Users\ecrist\SDRTrunk\SDRTrunk.properties]  [21MB/512MB 4%]
14:55:30.203 INFO  i.g.d.p.SystemProperties - SystemProperties - application properties loaded [C:\Users\ecrist\SDRTrunk\SDRTrunk.properties]  [21MB/512MB 4%]
14:55:30.388 INFO  i.g.d.s.t.m.TunerManager - Discovering tuners ...  [6MB/56MB 11%]
14:55:30.482 INFO  i.g.d.s.t.m.TunerManager - LibUsb API Version: 1.0.262  [7MB/56MB 12%]
14:55:30.483 INFO  i.g.d.s.t.m.TunerManager - LibUsb Version: 1.0.22.11312  [7MB/56MB 12%]
14:55:30.622 INFO  i.g.d.s.t.m.TunerManager - LibUsb - discovered [10] potential usb devices  [7MB/56MB 12%]
14:55:30.626 INFO  i.g.d.s.t.m.TunerManager - Discovered tuner at USB Bus [2] Port [11] Tuner Class [RTL-2832]  [7MB/56MB 13%]
14:55:30.628 INFO  i.g.d.s.t.m.TunerManager - Discovered tuner at USB Bus [2] Port [4] Tuner Class [RTL-2832]  [7MB/56MB 13%]
14:55:30.628 INFO  i.g.d.s.t.m.TunerManager - Discovered tuner at USB Bus [2] Port [12] Tuner Class [RTL-2832]  [7MB/56MB 13%]
14:55:30.629 INFO  i.g.d.s.t.m.TunerManager - Tuner: USB Tuner - RTL-2832 USB Bus:2 Port:11 - Added / Starting ...  [7MB/56MB 13%]
14:55:31.412 INFO  i.g.d.s.t.m.DiscoveredTuner - Tuner Error - Stopping - RTL-2832 USB Bus:2 Port:11 Error: Unsupported/Unrecognized Tuner Type  [9MB/56MB 16%]
14:55:31.413 INFO  i.g.d.s.t.m.DiscoveredTuner - Stopping Tuner: RTL-2832 USB Bus:2 Port:11  [9MB/56MB 16%]
14:55:31.414 ERROR i.g.d.s.t.m.DiscoveredUSBTuner - Unable to start tuner [RTL-2832] - error: Unsupported/Unrecognized Tuner Type  [9MB/56MB 16%]
14:55:31.418 INFO  i.g.d.s.t.m.TunerManager - Tuner: USB Tuner - RTL-2832 USB Bus:2 Port:4 - Added / Starting ...  [9MB/56MB 16%]
14:55:32.591 INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [10MB/56MB 19%]
14:55:32.593 INFO  i.g.d.s.t.m.TunerManager - Tuner: USB Tuner - RTL-2832 USB Bus:2 Port:4 - Applying Tuner Configuration  [10MB/56MB 19%]
14:55:32.801 INFO  i.g.d.s.t.m.TunerManager - Tuner: USB Tuner - RTL-2832 USB Bus:2 Port:12 - Added / Starting ...  [10MB/56MB 19%]
14:55:33.969 INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [11MB/56MB 19%]
14:55:33.970 INFO  i.g.d.s.t.m.TunerManager - Tuner: USB Tuner - RTL-2832 USB Bus:2 Port:12 - Applying Tuner Configuration  [11MB/56MB 19%]
14:55:34.176 INFO  i.g.d.s.t.m.TunerManager - LibUsb Hotplug event notification Is Not Supported on this platform.  [11MB/56MB 20%]
14:55:34.189 INFO  i.g.d.s.SettingsManager - SettingsManager - loading settings file [C:\Users\ecrist\SDRTrunk\settings\settings.xml]  [11MB/56MB 20%]
14:55:34.998 WARN  i.g.d.v.VectorUtilities - CPU supports maximum SIMD instructions of Species[float, 8, S_256_BIT]  [27MB/56MB 48%]
14:55:35.705 INFO  i.g.d.p.PlaylistManager - Loading playlist [C:\Users\ecrist\SDRTrunk\playlist\default.xml]  [18MB/56MB 33%]
14:55:36.007 INFO  i.g.dsheirer.gui.SDRTrunk - starting main application gui  [29MB/56MB 51%]
```

Here are the logs from my Mac:

```WARNING: Using incubator modules: jdk.incubator.vector
WARNING: package com.sun.java.swing.plaf.windows not in java.desktop
[... 18 lines cut ...]
15:26:37.292 INFO  i.g.d.log.ApplicationLog - Storage Directories:  [8MB/56MB 15%]
15:26:37.293 INFO  i.g.d.log.ApplicationLog -  Application Root: /Users/ecrist/SDRTrunk  [8MB/56MB 15%]
15:26:37.293 INFO  i.g.d.log.ApplicationLog -  Application Log:  /Users/ecrist/SDRTrunk/logs  [8MB/56MB 15%]
15:26:37.293 INFO  i.g.d.log.ApplicationLog -  Event Log:        /Users/ecrist/SDRTrunk/event_logs  [8MB/56MB 15%]
15:26:37.293 INFO  i.g.d.log.ApplicationLog -  Playlist:         /Users/ecrist/SDRTrunk/playlist  [9MB/56MB 16%]
15:26:37.294 INFO  i.g.d.log.ApplicationLog -  Recordings:       /Users/ecrist/SDRTrunk/recordings  [9MB/56MB 16%]
15:26:37.365 INFO  i.g.d.util.ThreadPool - Application thread pool created SCHEDULED and CACHED executors threads  [12MB/56MB 22%]
15:26:37.365 INFO  i.g.dsheirer.gui.SDRTrunk - Home path: /Users/ecrist/SDRTrunk  [12MB/56MB 22%]
15:26:37.366 INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [/Users/ecrist/SDRTrunk/SDRTrunk.properties]  [12MB/56MB 22%]
15:26:37.366 INFO  i.g.d.p.SystemProperties - SystemProperties - application properties loaded [/Users/ecrist/SDRTrunk/SDRTrunk.properties]  [12MB/56MB 22%]
15:26:37.419 INFO  i.g.d.s.t.m.TunerManager - Discovering tuners ...  [15MB/56MB 28%]
15:26:37.514 INFO  i.g.d.s.t.m.TunerManager - LibUsb API Version: 1.0.262  [15MB/56MB 28%]
15:26:37.515 INFO  i.g.d.s.t.m.TunerManager - LibUsb Version: 1.0.22.11312  [16MB/56MB 29%]
15:26:37.520 INFO  i.g.d.s.t.m.TunerManager - LibUsb - discovered [2] potential usb devices  [16MB/56MB 29%]
15:26:37.523 INFO  i.g.d.s.t.m.TunerManager - Discovered tuner at USB Bus [20] Port [2] Tuner Class [RTL-2832]  [16MB/56MB 29%]
15:26:37.525 INFO  i.g.d.s.t.m.TunerManager - Tuner: USB Tuner - RTL-2832 USB Bus:20 Port:2 - Added / Starting ...  [16MB/56MB 29%]
[... cut ...]
15:26:38.995 INFO  i.g.d.s.SettingsManager - SettingsManager - loading settings file [/Users/ecrist/SDRTrunk/settings/settings.xml]  [21MB/56MB 37%]
15:26:39.177 WARN  i.g.d.v.VectorUtilities - CPU supports maximum SIMD instructions of Species[float, 8, S_256_BIT]  [32MB/56MB 57%]
15:26:40.014 INFO  i.g.d.p.PlaylistManager - Loading playlist [/Users/ecrist/SDRTrunk/playlist/Itasca.xml]  [18MB/64MB 28%]
15:26:40.053 INFO  i.g.dsheirer.gui.SDRTrunk - starting main application gui  [21MB/64MB 33%]
```

Ctpolicelive, 2022-10-06:
Those are not the real RTL-SDR, they are cheap fake knock offs, note the period and not a dash, only buy the Official RTL-SDR's from their Amazon or ebay store [https://www.amazon.com/RTL-SDR-Blog-RTL2832U-Software-Defined/dp/B0129EBDS2/ref=asc_df_B0129EBDS2/?tag=hyprod-20&linkCode=df0&hvadid=309707619534&hvpos=&hvnetw=g&hvrand=5454729168136941627&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9003299&hvtargid=pla-760665313632&psc=1](url)

ecrist, 2022-10-07:
Doh.  They got me.  Returning.  This works fine with an official model I had laying around.

closed 2022-10-07 by ecrist