DSheirer/sdrtrunk #1323: PolyphaseChannelizer buffer overflows on machine with dual Xeon processors

tetrazocine, 2022-10-20:
Hello all, I have been trying to establish the reason for this app not running on my Lenovo workstation with dual Xeon processors. I have  P920 w/64gb ECC RAM and 2x Xeon Silver 4210's. Now this is more than sufficient I/O to run a single control channel on P25 Phase 1.

I have tested the exact same software (SDR Trunk x86_64-v0.5.0-beta5 and Java "19" 2022-09-20, Java(TM) SE Runtime Environment (build 19+36-2238)). On a Lenovo P53 Workstation Laptop with a Core i9 and 64gb RAM it performs without issue so seems to be something to do with the way the dual Xeon is processing the I/O potentially.

Using the Hetrodyne channelizer seems to work fine but with less channels of course. Any other similar experiences? I'll try running up a Windows 10 VM with one vCPU perhaps and see if this works and report back.

NB.

DSheirer, 2022-11-26:
@tetrazocine Any updates?

DSheirer, 2022-12-23:
Closing.  No updates.

closed 2022-12-23 by DSheirer