DSheirer/sdrtrunk #1325: Handling of patches on Harris systems

tvsjr, 2022-10-26:
Harris (maybe others?) systems use a different method of handling talkgroup "patching" (technically dynamic group regrouping) where, for each "patch" created, a supergroup is dynamically picked from a prespecified range, with the talkgroups being combined added under that supergroup. 

For a real-world example, the local sheriff's office has two primary patrol talkgroups, say GIDs 1 and 2. When things aren't busy, the two talkgroups will be combined under a single dispatcher. When this happens, a dynamic ID gets pulled from a range of (what appears to be) 65100-65199 and the supergroup gets built. Subscribers set on either GID 1 or GID 2 will hear the traffic from the "patch" (and this functions properly on subscribers from at least 3 different manufacturers).

Now, SDRTrunk will show only this supergroup ID and spits the same ID out via the log files (including the Trunking Recorder ingestion). This makes it impossible to assign an alias, enable/disable listening, etc.

The logic for handling these talkgroups needs to be adjusted to show the subgroups under the supergroup - both for display and for feeding to Trunking Recorder, etc.

DSheirer, 2022-10-27:
@tvsjr Could you create a bitstream recording of the control channel for that system, when you know that a patch group is being used, and note which identifiers are the patch group and which identifiers are the participating/patched groups.

I suspect that there's a Harris-specific message that they using that I need to add to the code.

tvsjr, 2022-10-28:
@DSheirer either I'm missing something incredibly obvious or something's broke. I enabled "Demodulated Bitstream (.bits)" under Recording on the Channels tab of the Playlist Editor. The app happily creates the file... but never puts anything in it. 0 bytes. I enabled the traffic channel bitstream - same issue.

What incredibly obvious thing am I missing here?

DSheirer, 2022-10-28:
There may be an issue.  I'll have to see what's going on with it.

DSheirer, 2022-11-26:
@tvsjr apologies ... the bitstream recording was broke and is now fixed with this PR: https://github.com/DSheirer/sdrtrunk/pull/1345

I'll be posting a new (final) 0.5.0 release shortly.  If you could make a bitstream recording of this Harris system once that's available?

DSheirer, 2023-11-11:
Closing.  Issue was resolved under separate issue number.

closed 2023-11-11 by DSheirer