DSheirer/sdrtrunk #1326: "USB Transfer Anomaly Detected" in sdrtrunk v0.5.0-beta5

bhall7, 2022-11-02:
I also posted this to the SDRTrunk email list, but thought it would be appropriate to log here:

I'm running SDR Trunk v0.5.0-beta5 on Windows 10 Pro (22H2), monitoring a new radio system that is in the process of being deployed ([UCA P25](https://www.radioreference.com/db/sid/12042)). Computer is an HP ProDesk 400 mini with an Intel 7th-Gen Core i5, 16GB RAM, and an NVMe SSD. SDR is the Airspy R2 plugged directly into the rear USB 3 port, with no other USB devices plugged in (running headless and accessing remotely via VNC).

After a while, I see the following error in the terminal window stating that a "USB transfer anomaly" was detected, after which I am not able to receive any traffic:

`08:08:15.634 ERROR i.g.d.s.t.u.USBTunerController - USB transfer anomaly detected - continuing - previous status [0 ] transferred [0] this has happened [1] times  [234MB/510MB 45%]
08:15:03.635 ERROR i.g.d.s.t.u.USBTunerController - USB transfer anomaly detected - continuing - previous status [0 ] transferred [0] this has happened [2] times  [162MB/510MB 31%]
08:16:53.233 ERROR i.g.d.s.t.u.USBTunerController - USB transfer anomaly detected - continuing - previous status [0 ] transferred [0] this has happened [3] times  [146MB/510MB 28%]
08:24:19.837 ERROR i.g.d.s.t.u.USBTunerController - USB transfer anomaly detected - continuing - previous status [0 ] transferred [0] this has happened [4] times  [343MB/510MB 67%]
08:25:22.433 ERROR i.g.d.s.t.u.USBTunerController - USB transfer anomaly detected - continuing - previous status [0 ] transferred [0] this has happened [5] times  [324MB/510MB 63%]
08:26:13.232 ERROR i.g.d.s.t.u.USBTunerController - USB transfer anomaly detected - continuing - previous status [0 ] transferred [0] this has happened [6] times  [346MB/510MB 67%]
08:28:50.435 ERROR i.g.d.s.t.u.USBTunerController - USB transfer anomaly detected - continuing - previous status [0 ] transferred [0] this has happened [7] times  [377MB/510MB 73%]
08:28:53.435 WARN  i.g.d.s.t.u.USBTunerController - USB transfer anomaly detection count exceeded the total number [8] of available tranfers - sdrtrunk may no longer be streaming data from this tuner and may appear to be locked up.  If so, restart the application to resolve the issue and please send the application log to the developer  [408MB/510MB 80%]`

After restarting the app, I see this error again, even after a full reboot. Although the radio system I'm monitoring is very quiet (still in the process of being deployed), SDRTrunk did log a group call, but never received the traffic (I have recording on for a wildcard alias with talkgroups 0-65,535).

Full application log attached.

Any idea what I can do to remediate this error?

Thanks!

[sdrtrunk_app.log](https://github.com/DSheirer/sdrtrunk/files/9924287/sdrtrunk_app.log)

bhall7, 2022-11-08:
More info on this: I switched to a different, live P25 Phase II radio system, which SDRTrunk monitors perfectly. But, I am still seeing the `ERROR i.g.d.s.t.u.USBTunerController - USB transfer anomaly detected` error in the terminal window.

I tried running with the Airspy R2 plugged in to each of the 6 USB ports on my machine. I also have the power management settings to "Performance" and USB power saving functions are all disabled.

It's worth mentioning that I have run this same hardware setup for years using Unitrunker/Trunking Recorder on a very busy, older Motorola Analog trunked system without any issues at all. I just recently started using SDRTrunk to listen to some newer P25 Phase II systems, and noticed the error. But, since it's just a "WARN" error, the program appears to continue running, although, I'm not very confident, given that it keeps popping up, no matter what I do. Should I just ignore it?

DSheirer, 2022-11-08:
For whatever reason, libusb is sending an empty buffer with no samples.  Not much sdrtrunk can do other than log the anomaly.  If you're not seeing any impact, then you might ignore the warning.

bhall7, 2022-11-08:
Cool, thanks, Denny!

closed 2022-11-08 by bhall7