DSheirer/sdrtrunk #1327: ERROR i.g.d.s.t.m.DiscoveredUSBTuner - Unable to start tuner [RTL-2832] - error: access denied

raul-parada, 2022-11-07:
Hi,

I'm using the version sdr-trunk-windows-x86_64-v0.5.0-beta5 on Windows 10. In the log appears this error:  `ERROR i.g.d.s.t.m.DiscoveredUSBTuner - Unable to start tuner [RTL-2832] - error: access denied - if using linux, blacklist the default driver and/or install udev rules`
Also in the Tuner tab appears "Error" and I cannot Restart tuner. The driver of the dongle works perfectly since it works on SDR sharp. On Zadig there are not interfaces available to assign drivers.

The dongle is this [one](https://www.google.com/url?sa=i&url=https%3A%2F%2Fwww.ebay.es%2Fitm%2F393485322076&psig=AOvVaw0ohuBbppOJoK2eLiuWHohf&ust=1667911381475000&source=images&cd=vfe&ved=0CA0QjhxqFwoTCOCuzZyMnPsCFQAAAAAdAAAAABAE)

Find attached the full log.
[log_sdrtrunk.txt](https://github.com/DSheirer/sdrtrunk/files/9951669/log_sdrtrunk.txt)

DSheirer, 2022-11-07:
This isn't a bug in the sdrtrunk software.  sdrtrunk is reporting that it is not able to access the dongle because your OS is preventing access and that's an issue you'll need to resolve.

You may need to check if there is some software application that is claiming the dongle and preventing sdrtrunk from using the dongle.

raul-parada, 2022-11-07:
I still get this error: Unrecognized embedded tuner type for RTL-2832

DSheirer, 2022-11-07:
That normally means that the tuner type is not supported by sdrtrunk ... it only supports R820T, R820T2, and E4000 tuners.

closed 2022-11-26 by DSheirer