DSheirer/sdrtrunk #1328: How to build and install 52-sdrplay-support

srs4511351, 2022-11-08:
I have sdrtrunk Version 0.5.0 Beta 5 running on a raspberry Pi 4.
The OS is DietPi 64 bit bookworm.
openjdk version "11.0.16" 2022-07-19
OpenJDK Runtime Environment (build 11.0.16+8-post-Debian-1)
OpenJDK 64-Bit Server VM (build 11.0.16+8-post-Debian-1, mixed mode)

I see that there is a 52-sdrplay-support branch and I would like to try it out with my RSP1A.

The wiki says to download and extract a zip archive, but I cannot find one for the 52-sdrplay-support branch.
I downloaded it, but the files needed to run it aren't there.
cd ~/
git clone https://github.com/DSheirer/sdrtrunk.git
cd sdrtrunk
git checkout 52-sdrplay-support

What needs to be done to get it running?

Thanks, Steve

scresante, 2022-11-24:
See the developer instructions in the README. It says to build it you have to have at least OpenJDK 17.

I have been trying to get this branch to build but have been unsuccessful and get build errors. I also get build errors with the master and 0.4.0 branch. I have openJDK 19 and the build fails both on ubuntu and arch.

srs4511351, 2022-11-24:
My second attempt was with JRE 17. The errors were different, but it still failed.
Please let me know if you get it to work.
There is a note somewhere and issues submitted that it only uses SDRplay API 2.
API 3.07 has been the standard for over a year now and a newer one is on the way. It is already available for Windows.

scresante, 2022-11-25:
@srs4511351  I gave up on sdrtrunk. My use case was to get a P25 SDR decoder working and that was it. I was able to get that working using boatbod's release + an ubuntu 20.04 VM. 

It would be very helpful if the maintainer of this project could release a Dockerfile that builds the release artifacts. @DSheirer : the build instructions seem to be missing something. I'm not familiar with gradle all too much but I've tried multiple combinations of JDK≥17 and OS's. What's your build process?

srs4511351, 2022-11-25:
I read that they bumped SDRplay support to the next release.
If they would use SoapySDR, it would open up support for a wide range of SDRs.
I have sdrtrunk running on my Raspberry Pi, but it is with a RTL-SDR. It's not mine, so I won't always have it. The CPU load is so high that I turn off the spectrum and waterfall display. It seems that Java is very CPU intensive.
I also use op25, boatbod build. It doesn't work with GNUradio > 3.8, so they will eventually have to support that. Their CPU load is very low.
I have my build procedure online. It shows you how to start from a Raspberry PI OS image and install a wide range of applications and utilities.
See: https://www.radiosrs.net/tech.html
If you go to my home page, you can see my collectibles, most;y radios.
https://www.radiosrs.net/

DSheirer, 2022-11-26:
Closing this as an issue.  Feel free to continue the conversation to Google groups or Discord.

I'm still developing on the #52 branch for the 0.6.0 milestone.  I'm trying to get 0.5.0 finished and released so that I can focus on sdrplay support again.

closed 2022-11-26 by DSheirer