DSheirer/sdrtrunk #1329: Issues importing aliases from radioreference.com

srs4511351, 2022-11-08:
With sdrtrunk Version 0.5.0 Beta 5, the import fields seem to be mixed up.

My system is Michigan's Public Safety Communications System (MPSCS)
The Description field on radioreference does not import to the Alias list. 
Of the descriptive fields, sdrtrunk aliases only uses the Group and the Alpha Tag from radioreference.
Alpha Tag is used as the Alias and is basically meaningless, so I have no idea who is talking.

The most meaningful, descriptive fields from radioreference are the Group and Description. If somehow both of these could appear in the Now Playing tab, it would be great!

Could a feature be added to change what is imported to which field? Adding the ability to display the Group in Now Playing would be very helpful.

Thanks, Steve

srs4511351, 2022-11-10:
If a feature is added to allow importing from a CSV or TSV file, I can provide what I want to see.

----Steve

SolderFish, 2023-02-20:
Slightly related I would love to see the option for Alpha tags to be based off the channel names.

schmitzna, 2023-12-20:
RadioReference Group + Description would make a great Alias for Trunking Recorder instead of the Alpha Tags.

SolderFish, 2025-02-12:
Following up for clarity on my earlier request, being able to set alias' based off the frequency.
This would be helpful for the Conventional P25 systems that aren't using the trunked system.