DSheirer/sdrtrunk #1332: Multi-level alias list

tvsjr, 2022-11-15:
I don't know if this applies elsewhere but, throughout Texas, system owners have bought into a state-run and delegated ID system. Ostensibly (this works probably 99% of the time), a single radio receives an ID via this scheme on a "home" system/agency. From there, if that radio needs comms on other systems, the sysadmin simply activates that same ID versus issuing a separate ID. It's a great idea. These IDs are usually all above 1M.

However, this means that, especially for assets that may have statewide comms capability (TxDPS helicopter, air ambulance services, etc.) the same ID might appear on more than a dozen P25 systems throughout the state. Which means tagging the same radio across multiple systems.

As a potential solution, consider adding a shared alias list (which could be for radio IDs only) that would be handled at a lower priority than the system-specific alias list. If an ID comes up, the app checks the system-specific alias list... if no match is found, it checks the shared list.

This would also make discovery of new, interesting stuff easier. If you determine that ID 5000001 is a DPS helicopter on one system, and the same ID shows up on another system, you have a very good idea of who it is.

k8tek, 2022-11-15:
Am I missing something or is this already possible?
[image]

k8tek, 2022-11-15:
[image]
Am I missing something or isn’t this already possible?

tvsjr, 2022-11-15:
You can reuse alias lists, yes… but there’s two problems with that. One, that means all aliases – including talkgroups. Since there aren’t enough possible group IDs, those do get reused across systems. Second, if a radio ID does get reused (there are systems who aren’t participating in the statewide ID plan yet or perhaps you’re near a border monitoring systems in two states where the ID system isn’t harmonized), you have the same overlap issue.

What I’m advocating is a tiered alias list strategy – which could even be limited to just radio IDs. Check the specific alias list for the system, then check a more general alias list that could be shared.

 

 <https://user-images.githubusercontent.com/69276965/201908443-27beebdf-54e6-4a1a-a0db-4d6bb7c1435f.png>

DSheirer, 2022-11-26:
I understand your need as you've described it, but unfortunately the solution would be fairly complex to develop.  Closing.

closed 2022-11-26 by DSheirer