DSheirer/sdrtrunk #1342: Is there a way to save a set of encrypted message ciphertext?

stancel, 2022-11-23:
Hi DSheirer,

I am interested in analyzing the ciphertext of some DES-OFB encrypted communications. Is there a way in SDRTrunk and/or using the JMBE library to save a sampling of such transmissions in order to analyze using different software at another time? Thanks in advance for your time and knowledge to answer.

DSheirer, 2022-11-25:
You can either turnon the bitstream recorder or you can turn on the logging of JMBE frames.

closed 2022-11-25 by DSheirer

stancel, 2022-11-27:
Thanks for the response. Very much appreciated. 

Please correct me if I am wrong on either or both of these methods (whenever you have a moment):

# 1 - Bitstream Recording
It looks like to turn on the bitstream recorder that I will just need to edit this line "mBitstreamRecorder.setEnabled(false);" in the src/main/java/io/github/dsheirer/record/RecordConfigurationEditor.java class. 

or

# 2 - JMBE Frame Logging 

It looks to turn on the logging of JMBE frames that I will just need to utilize the "mLog" variable ("private final static Logger mLog = LoggerFactory.getLogger(IMBEFrame.class);") in the "codec/src/main/java/jmbe/codec/imbe/IMBEFrame.java" of the JMBE library

and/or

"mLog" variable ("private final static Logger mLog = LoggerFactory.getLogger(AMBEFrame.class);") in the "codec/src/main/java/jmbe/codec/ambe/AMBEFrame.java" of the JMBE library

Thanks again for the time, help and your response. By the way, it is so nice to see well documented classes, methods, sanely named variables, etc. You have done an amazing job with both building SDRTrunk and the corresponding JMBE decoder library.