DSheirer/sdrtrunk #1349: [Feature Request] TETRA Decoding

adryd325, 2022-11-29:
I would really appreciate if decoding of (non-encrypted) TETRA was added.

adryd325, 2024-01-22:
From the messing around I did in my fork, the DQPSK demodulator for P25 could possibly be reworked a bit to work with TETRA. obviously 25KHz band width instead of 12.5. I definitely do not have the knowledge (or motivation) at this time to continue working on it however. 

If anyone is looking at my fork, please disregard it, I don't believe it would work at all with the direction I went. I remember going back a few months after having made it and realized most of it was nonsense

DSheirer, 2024-01-22:
I don't have access to any TETRA systems.  If someone wants to make some baseband control channel recordings using the FM demodulator set to 25 kHz I might have time in the near future to work on a demodulator for this.  If you'd like to contribute a few recordings, you can either post them here, or email them to me directly.  Denny

adryd325, 2024-01-22:
I have an IQ recording https://adryd.co/public/tetra_iq_recording.raw. I'll get you an FM recording or two shorly

DSheirer, 2024-01-22:
I need I/Q recordings made with sdrtrunk, using the FMNB demodulator with 25 khz channel filter.  sdrtrunk should produce a roughly 50kHz sample rate I/Q recording when you enable 'Baseband Recorder' in the recorder section of the channel configuration.

adryd325, 2024-01-22:
https://adryd.co/public/20240122_133458_412962500_Toronto_Transit_Commission_29_Toronto_Shipp_51_baseband.wav

JustApu, 2024-01-23:
Is this useful? https://drive.google.com/file/d/1QSrHbDKw6HmuCLMskXo_XBkDIDVf1-wu/view?usp=drive_link

If yes, I can dedicate an SDR to getting more samples.

adryd325, 2024-01-23:
> Is this useful? https://drive.google.com/file/d/1QSrHbDKw6HmuCLMskXo_XBkDIDVf1-wu/view?usp=drive_link
> 
> If yes, I can dedicate an SDR to getting more samples.

This is probably better than my recording, something is wrong with my sdrtrunk IQ recordings

DSheirer, 2024-01-23:
@JustApu Yes, I opened that recording with baudline and it looks like a nice strong signal.

DSheirer, 2024-01-23:
@adryd325 The 20240122_235425 file also looks like a good strong signal.

JustApu, 2024-01-24:
> @JustApu Yes, I opened that recording with baudline and it looks like a nice strong signal.

If you need additional data, please let us know. Not really a system I care to monitor but, it is in range, so I can certainly test.

ea7hjb, 2024-03-19:
Shall you need TETRA unencrypted samples of any kind (that is, a network with a single base station or a full trunked network with many base stations) I can provide, or even install a RTL_TCP/AirSpy/SDR++/whatever server for you.

DSheirer, 2024-03-24:
@ea7hjb Yes, can you please make a couple baseband recordings using the sdrtrunk software with the NBFM decoder set to 25kHz bandwidth?  If you can get examples of the different network configurations (single base, or full trunked) both with and with encryption, so that I can make sure I'm handling (for example ignoring) encrypted calls.

ea7hjb, 2024-03-28:
@DSheirer Hello. I made some recordings, and I could make many more, but I prefer not to upload them here. I don't know what is your email address. Could you please send me a mail to [mycallsign]@gmail.com?

adryd325, 2024-04-13:
https://adryd.co/public/20240413_005948_430050000_test_SITE_null_25_baseband.wav
 
TETRA in DMO (effectively simplex)

ea7hjb, 2024-08-21:
Any updates/hopes on this? :)

ghstonks, 2025-06-06:
Hi, I was wondering if anyone is still working on this feature. If needed I can supply samples or set up a remote desktop for development on this.