DSheirer/sdrtrunk #1350: TEARDOWN

Southernw2002, 2022-11-30:
Suddenly I have started to get lots of TEARDOWN messages, resulting in the programme coming to a halt... upgrading to the latest version does not seem to have changed the situation.

I dont think anything has change here.. so have no idea why it has started.. or if there is a way to fix.??

Any help would be appreciated

A

DSheirer, 2022-12-23:
@Southernw2002 is this still an issue?

Bits-Bytes1, 2024-07-21:
I am also experiencing teardown. I can submit system specs and whatever logs to help!

bushcraftbuddy, 2024-07-22:
> I am also experiencing teardown. I can submit system specs and whatever logs to help!

I have had Teardown issues for about a month now. I just noticed that my event logs were full so I deleted all the files and increased the Event log size in user preferences and have yet to have an issue 

if you need specs or any specifics, I can give them as needed

Bits-Bytes1, 2024-07-27:
I deleted the logs and that took care of the issue, thanks!

IMG20191023, 2024-08-03:
Still having teardown issue here, locking up channels.

Event log setting in preferences doesn't adjust the log size amount, per; https://github.com/DSheirer/sdrtrunk/issues/1823#issuecomment-1925724770

bushcraftbuddy, 2024-08-03:
Have you deleted the log files in the log directory?

Thanks for referencing that comment. I had not known about that but it seems that the log file size setting does still effect the application itself. Not just the bar at the bottom. When I had just deleted the log files I had Teardown within 30 minutes (P-25 Phase-1 with 47 Trunked Channel's) but when I also Changed the log preference I didn't have teardown. I'm at 15 days without teardown issues so something I did worked

From what I can tell It appears that when you get to a large log file it can't add anymore log entries so it fails to teardown. I would assume it has to do with some odd requirement with log file size with Windows

DSheirer, 2026-11-08:
This seems to be OBE, closing.

closed 2026-11-08 by DSheirer