DSheirer/sdrtrunk #1353: Polyphase Channelizer - Decode Channels Go IDLE After Period of Time

DSheirer, 2022-12-11:
When running multiple P25 control channels or other decoders, the channels go to an IDLE state after a period of time, indicating that they are no longer decoding the signal.

# Troubleshooting
I added a channel FFT display to the Now Playing - Channel tab to verify if the channel was either not receiving sample buffers or if the channel center tuned frequency was misaligned.  It turns out that each polyphase channel is misaligned by (approximately) FS/4.  This seems consistent when viewing each of the multiple channels that were active when the polyphase channelizer gets into this state.

[image]

# Task

Investigate what is causing the polyphase channelizer to misalign the channel sample streams.

DSheirer, 2022-12-16:
@cfc62 I think the issue may have been there since the initial addition of the Polyphase channelizer.  I think this is a threading issue.  When there are multiple control channels issuing concurrent requests for traffic channel allocations, I found that there's a chance for a disconnect between changing the tuner frequency and updating the channelizer frequency offsets where things can get out of whack.  I updated the channelizer to use thread locking to be more deliberate in the tuner and channelizer frequency updates and am testing that now to see if the issue is resolved.

cfc62, 2022-12-16:
Thanks for the update!

Brad-git-man, 2022-12-17:
@DSheirer Thanks for working on this issue. I run 5 SDRs in a fan cooled array and SDRT crashes after a while.

closed 2022-12-17 by DSheirer

DSheirer, 2022-12-17:
@Brad-git-man pull down the latest branch and see if this resolves the issue for you.  I ran for 24-hours continuous and didn't see the issue reappear .... it was popping up within 15 minutes to 3 hours for me previously.

Brad-git-man, 2022-12-17:
@DSheirer I went to the latest branch page, but can't figure out how to pull it down, is it a complete version or do you just get the new Channelizer there and replace yours with the new one?

DSheirer, 2022-12-18:
@Brad-git-man I'll be posting a new build shortly.

Brad-git-man, 2022-12-18:
@DSheirer I have now installed it and I really like the improvements, the audio now seems clearer, thanks for all your great work. You are moving ever closer to version 1.0 which I know will be a truly great program.

GTR8000, 2022-12-28:
0.5.0 Final, Win 10, RTL-2832/R820T, 2.4 Msps, auto-PPM enabled

I just experienced this IDLE condition for the first time that I can recall.

This install is decoding a single VHF P25 control channel, along with one "dummy" P25P1 channel that serves to keep the site's channels centered for optimal decoding. At 11:24 this morning it seems that the control channel decoder went into the idle state, according to the recordings in Trunking Recorder. There was activity 30 seconds after that last transmission which was never decoded (logged in Unitrunker), and it remained idle until I noticed it a few minutes ago.

Looking at the spectrum _(the channel FFT display mentioned in the first post is nowhere to be found)_, the dongle was functioning and was still within the frequency range of the control channel, as expected. I restarted the channel via the Playlist Editor, and it snapped right back to CONTROL status and is now decoding voice again.

Additional info: Well that's odd, I just looked again to verify that the tuner was set to Polyphase, which is how I always run SDRTrunk, and surprisingly it was set to Heterodyne. I have no way of knowing if I somehow set that myself a long time ago, or if it changed on its own somehow.

Unfortunately I'm therefore not able to determine whether the IDLE condition happened in Heterodyne or Polyphase mode. I've set it back to Polyphase and will continue to monitor.

PR #1354 (2022-12-17): #1353 Polyphase Channelizer Produces Idle Decode Channels After Period of Time
Closes #1353.

Resolves issue where polyphase channelizer would mis-tune channels after a period of time resulting in idle decoding channels.