DSheirer/sdrtrunk #1373: Broadcastify feeds constantly disconnecting (Version 0.5.0 Beta 7)

zacharyfleck, 2022-12-21:
All BCFY feeds are disconnecting on a regular basis. See attached image. This issue does not occur on earlier versions. This is not an internet issue.

@doug-hoffman - I experienced this when I tried to manually build your changes from your fork about a month ago. I'm wondering if perhaps it has something to do with the stream settings and how the non-silence is generated?

[image]

doug-hoffman, 2022-12-21:
FWIW, one of my feeds on beta 3 with #1244:

[image]

[image]

I've not used beta 7 in production yet. I'll give it a spin and see.

doug-hoffman, 2022-12-21:
16kbps/8kHz

[image]

zacharyfleck, 2022-12-21:
Ok. Mine varies from that. Wonder if that's somehow related. I'm at 32kbps/22050kHz

doug-hoffman, 2022-12-21:
My BCFY feeds have been stable on beta 7 for about 45 minutes.

I'm seeing issues with 32kbps/22.05kHz to a local icecast-2.4.4 server:

```
2022-12-21 12:15:26.624 ERROR i.g.d.a.b.AudioStreamingBroadcaster - Error while processing audio streaming queue  [925MB/1GB 70%]
java.lang.ArrayIndexOutOfBoundsException: Index 1 out of bounds for length 1
        at io.github.dsheirer.audio.convert.MP3Header.getMPEGVersion(MP3Header.java:60)
        at io.github.dsheirer.audio.convert.MP3Header.getFrameSamples(MP3Header.java:125)
        at io.github.dsheirer.audio.convert.MP3Header.getFrameDuration(MP3Header.java:157)
        at io.github.dsheirer.audio.convert.MP3AudioFrames.getCurrentFrameDuration(MP3AudioFrames.java:43)
        at io.github.dsheirer.audio.broadcast.AudioStreamingBroadcaster$RecordingQueueProcessor.run(AudioStreamingBroadcaster.java:302)
        at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:539)
        at java.base/java.util.concurrent.FutureTask.runAndReset(FutureTask.java:305)
        at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:305)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
        at java.base/java.lang.Thread.run(Thread.java:833)
```

32kbps/16kHz and 32kbps/8kHz seem fine.

I'll dig into this further this evening and see if I can figure out what's going on.

doug-hoffman, 2022-12-21:
I identified the cause of the error above. I believe that should also resolve your issue with BCFY.

PR #1375 (2022-12-23): Correctly identify padded MP3 frames
Resolves #1373.
closed 2022-12-23 by DSheirer