DSheirer/sdrtrunk #1374: Audio sounding "digitized" or "robotic" on new version?

bnelson333, 2022-12-21:
I tried poking around to see if I could find a post on this and didn't see anything, apologies if it's a duplicate. I upgraded to the latest version of trunk-recorder last week and ever since I have noticed a lot (but not all) of the audio clips saved sound "digitized" or "robotic". I thought it was my antennas or config or something but then I wondered if it was the version instead.

So this morning I rolled it back to the previous version (I was on 4.2.0) and sure enough, it seems to have solved the problem. So my question is - is this known? Is there something I need to do to fix for it in 4.4.0? Something in the config?

I believe this is the same dispatcher, so I have example audio files of the:

Before (4.2.0): https://sndup.net/j6kr/

and 

After (4.4.0): https://sndup.net/wntn/

Thoughts?

DSheirer, 2022-12-23:
Some recent changes in MP3 encoding may have produced what you're hearing.  The next release will include some MP3 encoder changes.

closed 2022-12-23 by DSheirer

bnelson333, 2022-12-23:
oh my god, I posted this on the wrong page, so sorry. This is a problem in trunk-recorder, not sdrtrunk!