DSheirer/sdrtrunk #1405: 0.5.0 final throwing out of memory errors

infecticide, 2023-01-12:
I have been working around issues with SDRTrunk crashing, stopping or doing odd things since the 0.4.x days.

At first I believe my issue was the low power laptop I was using not having the CPU power or adequate memory to monitor 2 P25 Phase 1 systems as I was getting lots of "TEARDOWN" hangs.

I have a crontab that kills the java process every 12 hours and restarts it.  This seems to have been working well with beta5 and 6.

Now it seems that no matter how much ram I throw at it, it isn't enough. I'm getting the error below.

VM Stats:
vCPUs: 4
Memory: 8GB
OS: Ubuntu 22.04 Jammy

Command line I use to launch SDRTrunk:
export DISPLAY=':0.0'
export SDR_TRUNK_OPTS='-Djava.awt.headless=true -Xms4096m -Xmx4096m'
/home/infecticide/sdr-trunk-linux-x64/bin/sdr-trunk > /home/infecticide/SDRTrunk/logs/sdrtrunk_console.log

2023-01-12 08:31:31.750 ERROR i.g.d.util.Dispatcher - Unexpected error thrown from the Dispatcher thread  [1GB/4GB 45%]
java.lang.OutOfMemoryError: unable to create native thread: possibly out of memory or process/resource limits reached
        at java.base/java.lang.Thread.start0(Native Method)
        at java.base/java.lang.Thread.start(Unknown Source)
        at io.github.dsheirer.util.Dispatcher.start(Dispatcher.java:110)
        at io.github.dsheirer.dsp.filter.channelizer.output.ChannelOutputProcessor.start(ChannelOutputProcessor.java:65)
        at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelSource.updateOutputProcessor(PolyphaseChannelSource.java:170)
        at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelSource.<init>(PolyphaseChannelSource.java:66)
        at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager.getChannel(PolyphaseChannelManager.java:180)
        at io.github.dsheirer.source.tuner.manager.PolyphaseChannelSourceManager.getSource(PolyphaseChannelSourceManager.java:447)
        at io.github.dsheirer.source.tuner.manager.TunerManager.getSource(TunerManager.java:501)
        at io.github.dsheirer.source.tuner.manager.TunerManager.getSource(TunerManager.java:427)
        at io.github.dsheirer.controller.channel.ChannelProcessingManager.startProcessing(ChannelProcessingManager.java:382)
        at io.github.dsheirer.controller.channel.ChannelProcessingManager.startChannelRequest(ChannelProcessingManager.java:318)
        at java.base/jdk.internal.reflect.DirectMethodHandleAccessor.invoke(Unknown Source)
        at java.base/java.lang.reflect.Method.invoke(Unknown Source)
        at com.google.common.eventbus.Subscriber.invokeSubscriberMethod(Subscriber.java:85)
        at com.google.common.eventbus.Subscriber$SynchronizedSubscriber.invokeSubscriberMethod(Subscriber.java:142)
        at com.google.common.eventbus.Subscriber.lambda$dispatchEvent$0(Subscriber.java:71)
        at com.google.common.util.concurrent.DirectExecutor.execute(DirectExecutor.java:31)
        at com.google.common.eventbus.Subscriber.dispatchEvent(Subscriber.java:68)
        at com.google.common.eventbus.Dispatcher$PerThreadQueuedDispatcher.dispatch(Dispatcher.java:109)
        at com.google.common.eventbus.EventBus.post(EventBus.java:267)
        at io.github.dsheirer.module.decode.p25.P25TrafficChannelManager.processPhase1ChannelGrant(P25TrafficChannelManager.java:374)
        at io.github.dsheirer.module.decode.p25.P25TrafficChannelManager.processChannelGrant(P25TrafficChannelManager.java:253)
        at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderState.processChannelGrant(P25P1DecoderState.java:333)
        at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderState.processAMBTCGroupVoiceChannelGrant(P25P1DecoderState.java:601)
        at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderState.processAMBTC(P25P1DecoderState.java:432)
        at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderState.receive(P25P1DecoderState.java:281)
        at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderState.receive(P25P1DecoderState.java:153)
        at io.github.dsheirer.sample.Broadcaster.broadcast(Broadcaster.java:151)
        at io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:50)
        at io.github.dsheirer.module.decode.Decoder$MessageDistributor.receive(Decoder.java:97)
        at io.github.dsheirer.module.decode.Decoder$MessageDistributor.receive(Decoder.java:90)
        at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageProcessor.receive(P25P1MessageProcessor.java:92)
        at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageProcessor.receive(P25P1MessageProcessor.java:34)
        at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageFramer.dispatchMessage(P25P1MessageFramer.java:273)
        at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageFramer.receive(P25P1MessageFramer.java:204)
        at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageFramer.receive(P25P1MessageFramer.java:70)
        at io.github.dsheirer.sample.Broadcaster.broadcast(Broadcaster.java:151)
        at io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:50)
        at io.github.dsheirer.dsp.psk.PSKDemodulator.broadcast(PSKDemodulator.java:61)
        at io.github.dsheirer.dsp.psk.DQPSKDecisionDirectedDemodulator.calculateSymbol(DQPSKDecisionDirectedDemodulator.java:88)
        at io.github.dsheirer.dsp.psk.PSKDemodulator.receive(PSKDemodulator.java:117)
        at io.github.dsheirer.dsp.psk.PSKDemodulator.receive(PSKDemodulator.java:93)
        at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderC4FM.receive(P25P1DecoderC4FM.java:110)
        at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderC4FM.receive(P25P1DecoderC4FM.java:40)
        at io.github.dsheirer.sample.Broadcaster.broadcast(Broadcaster.java:151)
        at io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:50)
        at io.github.dsheirer.source.tuner.channel.StreamProcessorWithHeartbeat.receive(StreamProcessorWithHeartbeat.java:118)
        at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelSource.receive(PolyphaseChannelSource.java:108)
        at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelSource.receive(PolyphaseChannelSource.java:41)
        at io.github.dsheirer.dsp.filter.channelizer.output.TwoChannelOutputProcessor.process(TwoChannelOutputProcessor.java:111)
        at io.github.dsheirer.dsp.filter.channelizer.output.ChannelOutputProcessor.lambda$new$0(ChannelOutputProcessor.java:50)
        at io.github.dsheirer.util.Dispatcher$Processor.run(Dispatcher.java:182)
        at java.base/java.lang.Thread.run(Unknown Source)

By the look of the memory stats displayed, it's only using 1 out of 4 GB of RAM that I assigned to it.

What can I do to assist in troubleshooting this?

Brad-git-man, 2023-01-12:
Yes I ave raising my ram in the SDR-Trunk.bat to 8 gigs for over a yr now (with just one 8 gig declaration and not two 4's. It did seem to help some before version final. I have 5 SDRs monitoring four P25 sites and a half dozen FMN freqs. And version final seems to not be very stable, I can only run SDRT for 20 minutes max, before it locks up and send my 16 core CPU to 100% load.

Ctpolicelive, 2023-01-12:
I've been running 0.5.0 Final 24/7 since it came out with 5 sdr's monitoring 3 p25 sites. Running a linux mint dedicated PC with 6 core AMD and 12gig ram, "8gig to usb's" and having no issues. Not having any lockup's, and not having any usb failures. Now I see the OP is running inside a VM, maybe that could be his issue? I'm not gloating or whatever,  just not having any issues. The final version is very stable for me.

DSheirer, 2023-01-15:
The error stack trace indicates that the operating system failed/refused to create a new native thread when Java asked for one.  It also indicates that this could be due to: possibly out of memory or process/resource limits reached

At a point in the 0.5.0 beta series, I changed from using a fixed-size thread pool to allocating a dedicated thread per decoding channel.  This significantly reduced the memory usage for users that were only decoding a couple channels, but it can increase memory usage for users decoding a large number of channels and it would increase the total native thread counts for use cases like yours, where you are processing a large number of channels.  A fixed-size thread pool is problematic because it can't scale up or scale down according to how you are using the application.

You may have to relook at the (thread/memory) resources you're allocating to your VM and to the JVM.  The application log entry that you posted indicates that sdrtrunk is only using 1G of the 4G fixed size memory that the JVM reserved from the OS.  There may be other constraints that the OS/VM are imposing on Java that's preventing this new thread creation. 

There's a number of articles discussing how to diagnose this issue if you search google: "java.lang.OutOfMemoryError: unable to create native thread"

infecticide, 2023-02-27:
So looking into this (I have moved on to 0.6.0-alpha3 since) 

The current instance of SDRT was started at 12:00pm today.

Using "ps -eLF | grep -c java" I see SDRT is currently using 864 / 1438 threads in use on the system.

"cat /proc/sys/kernel/threads-max shows" "6341"

It seems like there should be enough threads providing Java is closing them down once they are done.

3:20pm - 870 threads
3:25pm - 897 threads (+27)
3:30pm - 903 threads (+6)
3:35pm - 913 threads (+10)
3:40pm - 921 threads (+8)

20 mins = 51 threads
153 threads / hour
3672 threads / day

A fresh instance of SDRT uses about 85 threads.

There seems to be some resource creep going on.    

So, in less than 2 days Java would exhaust the max threads for the OS in my case.

infecticide, 2023-02-27:
Following a thread about using jstack to tracking down thread/memory leaks in Java, I produced 3 files each 5 mins apart, hopefully they are helpful.
[jstack_20230227_161500.txt](https://github.com/DSheirer/sdrtrunk/files/10844785/jstack_20230227_161500.txt)
[jstack_20230227_161000.txt](https://github.com/DSheirer/sdrtrunk/files/10844786/jstack_20230227_161000.txt)
[jstack_20230227_160500.txt](https://github.com/DSheirer/sdrtrunk/files/10844787/jstack_20230227_160500.txt)

infecticide, 2023-03-02:
Some more non-dev sysadmin analysis on my part:

I used this to analyze the jstack files:  https://jstack.review/

Seems like as time goes on more and more "sdrtrunk polyphase channel" threads are awaiting notification on "AbstractQueuedSynchronizer$ConditionObject"

I'm not a java dev so I have no idea what or where to start looking any further than this.

If there is anything else you would like me to try or capture, please let me know.

DSheirer, 2023-03-03:
@infecticide Thanks very much.  Something is preventing those traffic channel threads from shutting down.

Which version of sdrtrunk were you running when you ran the jstack thread dumps?

infecticide, 2023-03-03:
0.6.0-alpha3

DSheirer, 2023-03-04:
@infecticide I think I found the problem.  I created a test build for Linux x86-64 with the patch.  Could you download it and try it out and see if it resolves the issue with the thread leak.

https://drive.google.com/file/d/1_S4tPXxsh0852Hh3pXf9dEniWxf7Lems/view?usp=sharing

infecticide, 2023-03-04:
SDRT started out with 66 threads on a fresh start.

After running for about 10 mins I'm showing 70 threads and that number goes up and down depending on active calls.

Thanks Denny, I think you got it!

closed 2023-03-04 by infecticide

DSheirer, 2023-03-04:
No, thank you!  You found the clue I needed to figure out what was going on.  Thanks!

GTR8000, 2023-03-04:
I was experiencing this issue also, right up to the latest build of 0.6.0 alpha3, so I'm eager to build and test this. 👍

GTR8000, 2023-03-04:
Yikes...within a few seconds of running 0.6.0 alpha4, the main window locked up and it started throwing errors in the log...had to roll back to alpha3 to get it going again...

2023-03-04 15:25:42.955 WARN  i.g.d.util.Dispatcher - Temporary buffer overflow for thread [sdrtrunk polyphase buffer processor] - throwing away samples  [282MB/434MB 65%]
2023-03-04 15:25:52.962 WARN  i.g.d.util.Dispatcher - Temporary buffer overflow for thread [sdrtrunk polyphase buffer processor] - throwing away samples  [254MB/434MB 58%]
2023-03-04 15:26:02.973 WARN  i.g.d.util.Dispatcher - Temporary buffer overflow for thread [sdrtrunk polyphase buffer processor] - throwing away samples  [141MB/434MB 32%]
2023-03-04 15:26:12.981 WARN  i.g.d.util.Dispatcher - Temporary buffer overflow for thread [sdrtrunk polyphase buffer processor] - throwing away samples  [283MB/434MB 65%]
2023-03-04 15:26:22.990 WARN  i.g.d.util.Dispatcher - Temporary buffer overflow for thread [sdrtrunk polyphase buffer processor] - throwing away samples  [148MB/434MB 34%]
2023-03-04 15:26:32.999 WARN  i.g.d.util.Dispatcher - Temporary buffer overflow for thread [sdrtrunk polyphase buffer processor] - throwing away samples  [293MB/434MB 67%]
2023-03-04 15:26:43.009 WARN  i.g.d.util.Dispatcher - Temporary buffer overflow for thread [sdrtrunk polyphase buffer processor] - throwing away samples  [184MB/434MB 42%]
2023-03-04 15:26:53.021 WARN  i.g.d.util.Dispatcher - Temporary buffer overflow for thread [sdrtrunk polyphase buffer processor] - throwing away samples  [357MB/434MB 82%]

DSheirer, 2023-03-05:
@GTR8000 I published a V0.6.0-Alpha4 and a new V0.5.2(final) release with these changes.  Can you let me know if you're seeing the same issue with either of these releases?

GTR8000, 2023-03-06:
@DSheirer Just fired up the latest 0.6.0 alpha4 release, and so far so good. Did not see any of those **Temporary buffer overflow for thread** errors this time, will keep an eye on things.

GTR8000, 2023-03-06:
Updating the thread...crashed and burned again, logs emailed to Denny

PR #1477 (2023-03-05): #1405 Thread Leak in Polyphase Channelizer
Closes #1405 

Resolves issue with some polyphase channel threads not shutting down correctly after the tuner's center frequency changes and the polyphase channel changes from 1 > 2 channel slots, or vice-versa.  

Resolves issue with building distro releases using non-local JDKs.

Increases build to version 0.6.0 alpha 4