DSheirer/sdrtrunk #1432: crash

Allen-colton, 2023-01-29:
this is the crash dump from my install on kali lunix wasnt shore where to send them so im putting on here 

/home/candc/SDRTrunk/bin/hs_err_pid42701.log
/home/candc/SDRTrunk/bin/hs_err_pid3677.log
[hs_err_pid42701.log](https://github.com/DSheirer/sdrtrunk/files/10528797/hs_err_pid42701.log)
[hs_err_pid3677.log](https://github.com/DSheirer/sdrtrunk/files/10528798/hs_err_pid3677.log)

DSheirer, 2023-11-14:
Closing.  Please add a comment if this issue is not yet resolved and I'll reopen.

closed 2023-11-14 by DSheirer

Allen-colton, 2023-11-16:
It happens randomly i replaced my usb cables and it seems to of helped i
guess it flips it out if it looses connection with my hackrf

Colton King