DSheirer/sdrtrunk #1433: Status->"fade" - 5.1-beta1, also decode delay

geoffb1723, 2023-01-29:
Same Trunk system (NAC 909) P25 Phase 1 Simulcast, with 5.1-beta1, Status shows "Fade" on most calls, loses Control channel from time to time. With 5.0-beta5, no issues.  Looking at Channel Peak/Power, about the same using either releases.

Also I cant quite pin down that it seems to sometimes decode a call, sometimes not, but in any even not nearly as "realtime" , if that makes sense, as 5.0b-5, meaning that the audio is instantaneous on 5.0b5 but seems to take a second or two to begin playing. Sometimes no audio.

Brad-git-man, 2023-01-29:
Yeah, I'm having garbled audio in 5.1 Bets 1 and also when it tries to "retune" one of my 5 SDRs it fails with errors in the JVM "DOS Window"

DSheirer, 2023-11-14:
Closing.  Please add a comment if this issue is not yet resolved and I'll reopen.

geoffb1723, 2023-11-14:
thank D, all good. kind rgds

Closed #1433<https://github.com/DSheirer/sdrtrunk/issues/1433> as completed.