DSheirer/sdrtrunk #1435: Posting my SDRT 5.0 Beta 1 errors for the JVM Window

Brad-git-man, 2023-01-31:
I am posting my SDRT 5.0 Beta 1 errors for the JVM Window to see if the cause is my PC or SDRT. I have 5 SDRs and I still have 1 or 2 of them drop out after a few hours.

<details>
    <summary>View Here</summary>

Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException: Cannot read field "modelIndex" because "this.viewToModel[<parameter1>]" is null
        at java.desktop/javax.swing.DefaultRowSorter.convertRowIndexToModel(Unknown Source)
        at java.desktop/javax.swing.JTable.convertRowIndexToModel(Unknown Source)
        at java.desktop/javax.swing.JTable.getValueAt(Unknown Source)
        at java.desktop/javax.swing.JTable.prepareRenderer(Unknown Source)
        at java.desktop/javax.swing.plaf.basic.BasicTableUI.paintCell(Unknown Source)
        at java.desktop/javax.swing.plaf.basic.BasicTableUI.paintCells(Unknown Source)
        at java.desktop/javax.swing.plaf.basic.BasicTableUI.paint(Unknown Source)
        at java.desktop/javax.swing.plaf.ComponentUI.update(Unknown Source)
        at java.desktop/javax.swing.JComponent.paintComponent(Unknown Source)
        at java.desktop/javax.swing.JComponent.paint(Unknown Source)
        at java.desktop/javax.swing.JComponent.paintToOffscreen(Unknown Source)
        at java.desktop/javax.swing.RepaintManager$PaintManager.paintDoubleBufferedImpl(Unknown Source)
        at java.desktop/javax.swing.RepaintManager$PaintManager.paintDoubleBuffered(Unknown Source)
        at java.desktop/javax.swing.RepaintManager$PaintManager.paint(Unknown Source)
        at java.desktop/javax.swing.RepaintManager.paint(Unknown Source)
        at java.desktop/javax.swing.JComponent._paintImmediately(Unknown Source)
        at java.desktop/javax.swing.JComponent.paintImmediately(Unknown Source)
        at java.desktop/javax.swing.RepaintManager$4.run(Unknown Source)
        at java.desktop/javax.swing.RepaintManager$4.run(Unknown Source)
        at java.base/java.security.AccessController.doPrivileged(Unknown Source)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.desktop/javax.swing.RepaintManager.paintDirtyRegions(Unknown Source)
        at java.desktop/javax.swing.RepaintManager.paintDirtyRegions(Unknown Source)
        at java.desktop/javax.swing.RepaintManager.prePaintDirtyRegions(Unknown Source)
        at java.desktop/javax.swing.RepaintManager$ProcessingRunnable.run(Unknown Source)
        at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
        at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
        at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
        at java.base/java.security.AccessController.doPrivileged(Unknown Source)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)
2023-01-30 04:36:26.792 INFO  i.g.d.s.t.m.FrequencyErrorCorrectionManager - Auto-Correcting Tuner PPM to [-0.4]  [802MB/1GB 53%]
2023-01-30 04:47:22.912 INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2880000.0] providing [114] channels at [25263.2] Hz each  [828MB/1GB 54%]
2023-01-30 10:57:31.951 ERROR i.g.d.s.t.m.PolyphaseChannelSourceManager - Error getting source  [314MB/1GB 22%]
org.usb4java.LibUsbException: USB error 9: error writing byte buffer: Pipe error
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.write(RTL2832TunerController.java:798)
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.write(RTL2832TunerController.java:779)
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.writeI2CRegister(RTL2832TunerController.java:650)
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController$ControllerAdapter.writeI2CRegister(RTL2832TunerController.java:1550)
        at io.github.dsheirer.source.tuner.rtl.r820t.R820TEmbeddedTuner.writeR820TRegister(R820TEmbeddedTuner.java:487)
        at io.github.dsheirer.source.tuner.rtl.r820t.R820TEmbeddedTuner.setPLL(R820TEmbeddedTuner.java:389)
        at io.github.dsheirer.source.tuner.rtl.r820t.R820TEmbeddedTuner.setTunedFrequency(R820TEmbeddedTuner.java:148)
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.setTunedFrequency(RTL2832TunerController.java:142)
        at io.github.dsheirer.source.tuner.frequency.FrequencyController.setFrequency(FrequencyController.java:203)
        at io.github.dsheirer.source.tuner.frequency.FrequencyController.setFrequency(FrequencyController.java:175)
        at io.github.dsheirer.source.tuner.TunerController.setFrequency(TunerController.java:244)
        at io.github.dsheirer.source.tuner.manager.PolyphaseChannelSourceManager.getSource(PolyphaseChannelSourceManager.java:454)
        at io.github.dsheirer.source.tuner.manager.TunerManager.getSource(TunerManager.java:501)
        at io.github.dsheirer.source.tuner.channel.MultiFrequencyTunerChannelSource.getNextSource(MultiFrequencyTunerChannelSource.java:117)
        at io.github.dsheirer.source.tuner.channel.MultiFrequencyTunerChannelSource.rotate(MultiFrequencyTunerChannelSource.java:102)
        at io.github.dsheirer.source.tuner.channel.MultiFrequencyTunerChannelSource.process(MultiFrequencyTunerChannelSource.java:309)
        at io.github.dsheirer.source.SourceEventListenerToProcessorAdapter.receive(SourceEventListenerToProcessorAdapter.java:51)
        at io.github.dsheirer.source.SourceEventListenerToProcessorAdapter.receive(SourceEventListenerToProcessorAdapter.java:25)
        at io.github.dsheirer.sample.Broadcaster.broadcast(Broadcaster.java:151)
        at io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:50)
        at io.github.dsheirer.source.tuner.channel.rotation.ChannelRotationMonitor.checkState(ChannelRotationMonitor.java:148)
        at io.github.dsheirer.source.tuner.channel.rotation.ChannelRotationMonitor.lambda$start$0(ChannelRotationMonitor.java:179)
        at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
        at java.base/java.util.concurrent.FutureTask.runAndReset(Unknown Source)
        at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
        at java.base/java.lang.Thread.run(Unknown Source)
</details>

DSheirer, 2023-07-14:
First error with NullPointerException resolved under PR #1601

DSheirer, 2023-11-14:
Closing.  Please add a comment if this issue is not resolved and I'll reopen.

closed 2023-11-14 by DSheirer