DSheirer/sdrtrunk #1439: Very noticable squelch tail for NBFM frequencies, adding of CTCSS + DCS tones requested

dc1792, 2023-02-04:
I've been using SDR-trunk to broadcast my county's analog public safety system to broadcastify and I love the setup and user interface of the software. But there are some issues I've stumbled into. The squelch selectors are very sensitive and if not set specifically, it will cause open squelch to sound after and sometimes even before a transmission. Another thing that might help with this might be adding in CTCSS and DCS tone selector capabilities which might limit the squelch tail completely. Maybe adding this in the Channel bar, next to the frequency, in the playlist selection tab, would make the most sense. I'm not an expert, just some issues I've experienced and resolutions I think are appropriate. Thanks!

DSheirer, 2023-06-10:
@dc1792 Is this still an issue with the latest 0.6.0 alpha 6 release?  I changed the NBFM squelch to use a power squelch which should mostly eliminate any static bursts at the beginning or end of the transmissions.

jpryor223, 2023-06-11:
Static bursts for NBFM decoder remain present in the latest 0.6.0 alpha 6 release.  The beginning of transmissions are very brief, but loud enough to be a distraction, the end of transmissions are sometimes brief, sometimes a bit longer and louder sounding when listening.  Finding some times a recording has a static burst in the middle when the repeater is quickly dropping and coming back up.  Three examples here in screen shots of recent recordings, the static bursts in yellow.  The items in blue are MDC signaling.  From my listening the yellow beginning and ending static bursts are consistently present.  Setting the current power squelch as high as possible can help keep the ending static about as short as possible, just still loud and noticeable enough to be more distracting than a traditional radio squelch.

Will be interested to see if the CTCSS and/or DCS squelch can help clean the remainder of the NBFM squelch static noise.

[image]

krtz07, 2023-09-06:
I can confirm this issue - using the version released last week. I've listened around 160.92 MHZ and the sudden burst of static occurs. FM at 12.5bw.

kbroderick, 2023-10-07:
I'm also having difficulty getting a combination of squelch and gain settings that consistently works while monitoring a traditional NBFM system; I expect being able to filter based on CTCSS tones would help dramatically.

I suspect the underlying issue is a combination of multiple repeaters and my relative proximity to some but not all, along with the mobile units contacting the repeaters with varying signal strength, so I hope that a CTCSS decode should allow me to run a less-aggressive squelch cutoff without getting random noise.

grahampaull, 2024-09-12:
This would be a very welcome addition.

Repeaters are fine for me, with them having a constant power output I am able to turn off Auto Track and set the SQ manually, this limits the Squelch Tail as I am able to turn up the SQ right on the edge.

Listening to Marine Band for example I have to turn on Auto Track as signals are variable, stronger signals cause a longer SQ tail than Weaker Signals as the SQ slowly drops down to the noise floor, weaker signals SQ doesn't have as far to drop to the noise floor.

Having CTCSS would be nice, but for non-toned comms like Marine Band, having a quicker SQ would shorten the SQ Tail significantly.

N6MIK, 2025-06-05:
Bringing this back to life - wanted to throw my request in for DPL/DCS decoding as well - I push 7 busy, analog conventional channels to Trunking Recorder that have CWID set to 'try again' until not interrupted by voice traffic. This means that it's very possible to get multiple tries at the CWID every 15 mins. The IDer doesn't have DCS, but it is recorded none the less. Also - I'm having the long pre- and post- transmission squelch bursts on these channels (0.6.1 Final, and the latest Nightly).  I am hopeful  that DCS decode could help with those too! Thanks!

closed 2026-09-26 by DSheirer