DSheirer/sdrtrunk #1446: White screen on launch

Josh-SMU, 2023-02-13:
Linux aarch64 builds (0.5.1 beta1 and 0.5.0 final) on Debian Bookworm, 4GB RAM, Wayland

When I launch sdr-trunk it just hits a blank white screen. There is a popup for CPU calibration but even if I do that (the settings window it opens works correctly), once I close the settings window to return to the main application window it is still blank white. No errors in the console or app log.

gsruff, 2023-02-14:
I had this same problem when using dwm on Arch Linux. What solved it for me was using the wmname suckless utility to set my window manager to LG3D.

See these pages for more info:

- https://tools.suckless.org/x/wmname/
- https://wiki.archlinux.org/title/java#Gray_window,_applications_not_resizing_with_WM,_menus_immediately_closing
- https://wiki.archlinux.org/title/java#Impersonate_another_window_manager

Josh-SMU, 2023-02-14:
Awesome, thanks! Is there a way to do this for just one application (like passing it as an environmental variable) or does it only work globally?

Josh-SMU, 2023-02-14:
Ah nevermind, the variable wasn't set anyway. This works now, thank you!

closed 2023-02-14 by Josh-SMU

gsruff, 2023-02-14:
A simple bash script could set it, launch the specific application, then reset it on exit:
````
#!/usr/bin/bash
wmname LG3D
/path/to/sdrtrunk
wmname foo # replace foo with original value
````