DSheirer/sdrtrunk #1459: Fatal error when trying to start v0.6.0-alpha2

Brad-git-man, 2023-02-19:
How do I fix this fatal error when trying to start v0.6.0-alpha2
[image]

uzbek197, 2023-02-19:
Такая же ошибка НЕ ЗАПУСКАЕТСЯ ЧТО ДЕЛАТЬ?????

DSheirer, 2023-02-20:
This error means that you're using an older version of Java than what is needed.

Your error indicates that you're trying to run Version 0.6.0-alpha2.  Are you running the published release for that version, using the Java that is bundled with the release, or have you pulled down the source code and you're attempting to run it using a locally installed version of Java?

closed 2023-02-20 by DSheirer

uzbek197, 2023-02-20:
Загрузил версию 0.6.0-alpha3 и все заработало СПАСИБО БОЛЬШОЕ Отличная
работа!!!

uzbek197, 2024-11-22:
Привет Всем возник такой вопрос использую
sdr-trunk-windows-x86_64-v0.6.0-beta-5 в папке recordings появился файл
LST_XXX.dat и вот такой вопрос что это за файл и как его можно
посмотреть????
Всем заранее спасибо!!!

PR #1463 (2023-02-20): #1459 Gradle Doesn't Produce Correct OS Releases in Version 0.6.0
Closes #1459 Updates gradle build to fix bug related to building releases for each OS