DSheirer/sdrtrunk #1460: Java not founjd fatal Error in -v0.5.1 final

Brad-git-man, 2023-02-19:
[image]

I am getting a fatal error that it can't find Java. I am running the x86-64 version of SDRT on Windows 10 and no prior versions of SDRT has had this error on my computer.

DGxInfinitY, 2023-02-19:
Having the Same Problem in the Linux AArch64 Package in V0.5.1 Final.

DSheirer, 2023-02-20:
This error usually means that you're using an older version of Java than what is needed.

Your error description indicates that you're trying to run Version 0.5.1. Are you running the published release for that version, using the Java that is bundled with the release, or have you pulled down the source code and you're attempting to run it using a locally installed version of Java?

Version 0.5.1 requires JDK 18.

Brad-git-man, 2023-02-20:
@DSheirer I get this Java error where trying to start the regular download version of v0.5.1 final with no modifications. As I said above its the Windows x86-64 version on Windows 10. I have been downloading your version releases for 2 years and none of them have had any fatal Java errors and they just stared & ran fine.

DSheirer, 2023-02-20:
@Brad-git-man thanks!  I'll close this issue.

closed 2023-02-20 by DSheirer