DSheirer/sdrtrunk #1481: SDRPlay RSP1 - Hetrodyne Failure and Tuner Config Error

ScanCaster, 2023-03-10:
Version : 0.6.0b4  - (Also 0.6.0b3)
OS: *bunutu based 20.04 & 22.04's
SDRPlay API 3.0.7
Device: RSP1

Issue 1 = Change Tuner Processor from Polyphase to Hetrodyne upon restart you get the following:

2023-03-09 21:01:21.394 ERROR i.g.d.s.t.c.TunerConfigurationManager - Error loading tuner configuration file  [18MB/56MB 33%]
com.fasterxml.jackson.databind.exc.InvalidTypeIdException: Could not resolve type id 'Rsp1TunerConfiguration' as a subtype of `io.github.dsheirer.source.tuner.configuration.TunerConfiguration`: known type ids = [airspyTunerConfiguration, e4KTunerConfiguration, fcd1TunerConfiguration, fcd2TunerConfiguration, hackRFTunerConfiguration, r820TTunerConfiguration, recordingTunerConfiguration, rsp1aTunerConfiguration, rsp2TunerConfiguration, rspDuoTuner1Configuration, rspDuoTuner2Configuration, rspDxTunerConfiguration] (for POJO property 'tunerConfigurations')
 at [Source: (File); line: 4, column: 14] (through reference chain: io.github.dsheirer.source.tuner.configuration.TunerConfigurationState["tunerConfigurations"]->java.util.ArrayList[0])
        at com.fasterxml.jackson.databind.exc.InvalidTypeIdException.from(InvalidTypeIdException.java:43)
        at com.fasterxml.jackson.databind.DeserializationContext.invalidTypeIdException(DeserializationContext.java:2078)
        at com.fasterxml.jackson.databind.DeserializationContext.handleUnknownTypeId(DeserializationContext.java:1569)
        at com.fasterxml.jackson.databind.jsontype.impl.TypeDeserializerBase._handleUnknownTypeId(TypeDeserializerBase.java:298)
        at com.fasterxml.jackson.databind.jsontype.impl.TypeDeserializerBase._findDeserializer(TypeDeserializerBase.java:165)
        at com.fasterxml.jackson.databind.jsontype.impl.AsPropertyTypeDeserializer._deserializeTypedForId(AsPropertyTypeDeserializer.java:125)
        at com.fasterxml.jackson.databind.jsontype.impl.AsPropertyTypeDeserializer.deserializeTypedFromObject(AsPropertyTypeDeserializer.java:110)
        at com.fasterxml.jackson.databind.deser.AbstractDeserializer.deserializeWithType(AbstractDeserializer.java:263)
        at com.fasterxml.jackson.databind.deser.std.CollectionDeserializer._deserializeFromArray(CollectionDeserializer.java:361)
        at com.fasterxml.jackson.databind.deser.std.CollectionDeserializer.deserialize(CollectionDeserializer.java:244)
        at com.fasterxml.jackson.databind.deser.std.CollectionDeserializer.deserialize(CollectionDeserializer.java:28)
        at com.fasterxml.jackson.databind.deser.impl.MethodProperty.deserializeAndSet(MethodProperty.java:129)
        at com.fasterxml.jackson.databind.deser.BeanDeserializer.vanillaDeserialize(BeanDeserializer.java:314)
        at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserialize(BeanDeserializer.java:177)
        at com.fasterxml.jackson.databind.deser.DefaultDeserializationContext.readRootValue(DefaultDeserializationContext.java:323)
        at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:4730)
        at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:3542)
        at io.github.dsheirer.source.tuner.configuration.TunerConfigurationManager.load(TunerConfigurationManager.java:83)
        at io.github.dsheirer.source.tuner.configuration.TunerConfigurationManager.<init>(TunerConfigurationManager.java:66)
        at io.github.dsheirer.source.tuner.manager.TunerManager.<init>(TunerManager.java:88)
        at io.github.dsheirer.gui.SDRTrunk.<init>(SDRTrunk.java:177)
        at io.github.dsheirer.gui.SDRTrunk.main(SDRTrunk.java:738)

$ cat tuner_configuration.json 
{
  "disabledTuners" : [ ],
  "tunerConfigurations" : [ {
    "type" : "Rsp1TunerConfiguration",
    "sampleRate" : "RATE_8_000",
    "gain" : 14,
    "uniqueID" : "RSP1 SER#X000XX000X",
    "frequency" : 101100000,
    "frequencyCorrection" : 0.0,
    "autoPPMCorrectionEnabled" : true
  } ]

DELETING THIS FILE and restarting and this error goes away. 

Issue 2 - NO Tuning via Hetrodyne tuner module with RSP1, you get the following error when you attempt to START A CHANNEL from the playlist. Tuner Tab  shows 1 LOCKED, and the Now Playing shows the channel tuned, but is IDLE, there is NO CC LOCK. 

2023-03-09 21:01:45.761 ERROR i.g.d.m.ProcessingChain - Error starting module  [76MB/140MB 54%]
java.lang.ArithmeticException: / by zero
        at io.github.dsheirer.source.tuner.manager.HeterodyneChannelSourceManager.startDelayBuffer(HeterodyneChannelSourceManager.java:265)
        at io.github.dsheirer.source.tuner.manager.HeterodyneChannelSourceManager$ChannelSourceEventProcessor.receive(HeterodyneChannelSourceManager.java:297)
        at io.github.dsheirer.source.tuner.manager.HeterodyneChannelSourceManager$ChannelSourceEventProcessor.receive(HeterodyneChannelSourceManager.java:287)
        at io.github.dsheirer.source.tuner.channel.TunerChannelSource.broadcastProducerSourceEvent(TunerChannelSource.java:187)
        at io.github.dsheirer.source.tuner.channel.TunerChannelSource.start(TunerChannelSource.java:102)
        at io.github.dsheirer.source.tuner.channel.HalfBandTunerChannelSource.start(HalfBandTunerChannelSource.java:93)
        at io.github.dsheirer.module.ProcessingChain.start(ProcessingChain.java:726)
        at io.github.dsheirer.controller.channel.ChannelProcessingManager.startProcessing(ChannelProcessingManager.java:516)
        at io.github.dsheirer.controller.channel.ChannelProcessingManager.start(ChannelProcessingManager.java:303)
        at io.github.dsheirer.gui.playlist.channel.ChannelConfigurationEditor.lambda$getPlayButton$3(ChannelConfigurationEditor.java:372)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
        at java.base/java.lang.Thread.run(Unknown Source)

To use the SDRP - RSP1, change back to the Poly tuner, then the first error is there. 

The first error has been persistent since accessing Tuner section of the preferences, until you delete the tuner_configuration.json file. Then it goes away. It will RETURN if you access Tuner and change to the tuner processor.

DSheirer, 2023-03-11:
The sdrplay configuration changes from version 0.6.x are not backward compatible with 0.5.x.  The errors you're seeing happen when you first run 0.6.0 and the config gets updated with sdrplay tuner info, and then you run version 0.5.x software that has no awareness of sdrplay.

As you discovered, deleting the configuration file so that sdrtrunk can recreate it is the best approach here.

closed 2023-03-11 by DSheirer

ScanCaster, 2023-03-11:
**_Sorry, but that is NOT the case._**

060b3 and then 060b4 were tried on my main testing machine as its more CPU heft than my main use. 

060b3 was RUN.... tested RSP1,OK... working... change to HETRODYNE SETTING for tuner. 

You get the FIRST MESSAGE. I can RINSE, LATHER, REPEAT this... Deleting the tuner file after the change of tuner modes, and it doesn't appear. Leave that tuner file after a mode, poly/hetro, change and that error shows up. 060x branch. b3 or b4, doesn't matter. 

I can RINSE, LATHER, REPEAT on DragonOSX 29.x 2/22/23 which has **_NEVER HAD 050x series installed_**, matter of fact its a BRAND NEW INSTALL of the ISO to a VM to test out the distro for future SDR boxen.

Just tried this.... plugged in the RSP1, started up SDRT 0.6.0.b4. Change from Poly to Hetro in tuner tab, restart 0.6.0.b4, ERROR just as posted.

I can 100% replicate this on every access of the tuner tab to change from poly/hetro, with out fail

And the second error remains. Hetro will FAIL to start on the RSP1.

If I get a change I can put a video together of this, its 100% reproducible every time. Change the tuner mode in the tuner settings, and the error appears. 060x series b4 is being used, but b3 will do it too.

The DragonOSX VM has never even had anything below 060b3 on it, as again, its a new VM I used for testing to determine many things... in re RSP setup among others.

ScanCaster, 2023-03-11:
Additionally the DragonOSX VM has a _**BRAND NEW CLEAN setup. No files were transferred, ie: no playlists, or anything.**_ Only thing I scp'd over was the JMBE.jar as its not on the ISO.... I recreated by hand the channels I wanted to test the setup with... 

**_Error started on the DrgaonOSX VM_** and then I moved to the other testing machine to see what was going on... As well as test on bare metal with out the VM and USB HAL of the VM getting in the way of things.

This is 100% every time reproducible on the DragaonOSX with any change in the tuner setting from/to poly/hetro.  As well as the fail to start error, and nothing decoded.