DSheirer/sdrtrunk #1494: Hello im new to this

discobiscuit65, 2023-03-15:
My problem is when I try to set up playlist editor and click new it will not let me pick any decoders 
[image]
       
Please help Thanks

DSheirer, 2023-03-15:
When you click the New button, does it show a drop down list of decoders or what does it show?

discobiscuit65, 2023-03-15:
when I click it no it does not show nothing at all

bradjtrammell, 2023-06-23:
I realize this was posted months ago, and this is kind of a stupid question... but do you have the proper drivers installed?

Linux (except DragonOS Focal) doesn't ship by default with the SDR Libraries needed to operate most SDR Dongles. You need to install them from the command line.

Run `sudo apt-get install rtlsdr`, then reboot and launch SDRTrunk again. That SHOULD enable you to be able to control the dongle from software.

discobiscuit65, 2023-07-21:
I do thanks for the help iv got it working just a simple fix it was a mistake on my part

DSheirer, 2023-07-22:
Closing as resolved/OBE.

closed 2023-07-22 by DSheirer