DSheirer/sdrtrunk #1507: Cap+ - Incorrect LCN/LSN numbers

wedgef5, 2023-03-24:
The LCN/LSN as seen in the decoded messages pane appear to be off. While monitoring "repeater 1" on a local Cap+ system, I will see LCN=1 LSN=2 and LCN=2 LSN=3. "Repeater 2" will have LCN=2 LSN=4 and LCN=3 LSN=5. "Repeater 3" will have LCN=3 LSN=6 and LCN=4 LSN=7. There are only 3 frequencies on the site, so LCN=4 does not make sense. Also, I have never seen LCN=1 LSN=1 while monitoring "repeater 1" for an extended period of time.

cschmittiey, 2023-04-02:
a couple of us in the sdrtrunk discord have also noticed this - probably worth looking into. @DSheirer Is there anything we can provide to help debug?

wedgef5, 2023-04-19:
Watching SRDtrunk and DSD+-FL side-by-side and SDRtrunk's LSN is consistently +1 from what DSD+ shows.

wedgef5, 2023-04-22:
I believe I have identified the problem with this. I’m working on potential fixes.

DSheirer, 2023-05-19:
Closed via MR #1536