DSheirer/sdrtrunk #1508: Channel Teardown Error Leaves Dead Channels in Now Playing

DSheirer, 2023-03-25:
When an error occurs during shutdown of a traffic channel, that channel remains in the Now Playing table and the user has no options (other than restart) to clean out the traffic channel.

Add right-click context menu item to allow the user to remove the offending entry.  More than likely the traffic channel backing the table row no longer exists, but at least the UI cleanup can happen.

msgerbs, 2023-04-19:
I'm on 5.0.3 and had this happen again today. I had this error in my log but I have no idea when this happened so I don't know if it's related or not, sorry.

`20230419 134201.723 [sdrtrunk polyphase channel thread 1] ERROR i.g.d.d.f.c.PolyphaseChannelSource - Error dispatching complex samples to listener [class io.github.dsheirer.sample.Broadcaster]  [574MB/1GB 33%]
java.lang.Error: Interrupted attempt to acquire write lock
	at java.desktop/javax.swing.text.AbstractDocument.writeLock(Unknown Source)
	at java.desktop/javax.swing.text.AbstractDocument.replace(Unknown Source)
	at java.desktop/javax.swing.text.JTextComponent.setText(Unknown Source)
	at io.github.dsheirer.gui.control.JFrequencyControl$Digit.set(JFrequencyControl.java:417)
	at io.github.dsheirer.gui.control.JFrequencyControl$Digit.setFrequency(JFrequencyControl.java:295)
	at io.github.dsheirer.gui.control.JFrequencyControl.setFrequency(JFrequencyControl.java:156)
	at io.github.dsheirer.source.tuner.ui.TunerEditor$FrequencyPanel.updateControls(TunerEditor.java:631)
	at io.github.dsheirer.source.tuner.ui.TunerEditor.setTunerLockState(TunerEditor.java:553)
	at io.github.dsheirer.source.tuner.rtl.r820t.R820TTunerEditor.setTunerLockState(R820TTunerEditor.java:436)
	at io.github.dsheirer.source.tuner.ui.DiscoveredTunerEditor.setTunerLockState(DiscoveredTunerEditor.java:69)
	at io.github.dsheirer.source.tuner.ui.TunerViewPanel.lambda$init$3(TunerViewPanel.java:174)
	at io.github.dsheirer.source.tuner.ui.DiscoveredTunerModel.broadcast(DiscoveredTunerModel.java:296)
	at io.github.dsheirer.source.tuner.ui.DiscoveredTunerModel.receive(DiscoveredTunerModel.java:383)
	at io.github.dsheirer.source.tuner.ui.DiscoveredTunerModel.receive(DiscoveredTunerModel.java:43)
	at io.github.dsheirer.sample.Broadcaster.broadcast(Broadcaster.java:124)
	at io.github.dsheirer.source.tuner.Tuner.broadcast(Tuner.java:281)
	at io.github.dsheirer.source.tuner.Tuner.process(Tuner.java:189)
	at io.github.dsheirer.source.tuner.frequency.FrequencyController.broadcast(FrequencyController.java:367)
	at io.github.dsheirer.source.tuner.frequency.FrequencyController.setSampleRateLocked(FrequencyController.java:103)
	at io.github.dsheirer.source.tuner.TunerController.setLockedSampleRate(TunerController.java:219)
	at io.github.dsheirer.source.tuner.manager.PolyphaseChannelSourceManager.process(PolyphaseChannelSourceManager.java:500)
	at io.github.dsheirer.sample.Broadcaster.broadcast(Broadcaster.java:124)
	at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager.startChannelSource(PolyphaseChannelManager.java:227)
	at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$ChannelSourceEventListener.receive(PolyphaseChannelManager.java:422)
	at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$ChannelSourceEventListener.receive(PolyphaseChannelManager.java:412)
	at io.github.dsheirer.source.tuner.channel.TunerChannelSource.broadcastProducerSourceEvent(TunerChannelSource.java:186)
	at io.github.dsheirer.source.tuner.channel.TunerChannelSource.start(TunerChannelSource.java:102)
	at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelSource.start(PolyphaseChannelSource.java:107)
	at io.github.dsheirer.module.ProcessingChain.start(ProcessingChain.java:726)
	at io.github.dsheirer.controller.channel.ChannelProcessingManager.startProcessing(ChannelProcessingManager.java:516)
	at io.github.dsheirer.controller.channel.ChannelProcessingManager.startChannelRequest(ChannelProcessingManager.java:318)
	at java.base/jdk.internal.reflect.DirectMethodHandleAccessor.invoke(Unknown Source)
	at java.base/java.lang.reflect.Method.invoke(Unknown Source)
	at com.google.common.eventbus.Subscriber.invokeSubscriberMethod(Subscriber.java:85)
	at com.google.common.eventbus.Subscriber$SynchronizedSubscriber.invokeSubscriberMethod(Subscriber.java:142)
	at com.google.common.eventbus.Subscriber.lambda$dispatchEvent$0(Subscriber.java:71)
	at com.google.common.util.concurrent.DirectExecutor.execute(DirectExecutor.java:31)
	at com.google.common.eventbus.Subscriber.dispatchEvent(Subscriber.java:68)
	at com.google.common.eventbus.Dispatcher$PerThreadQueuedDispatcher.dispatch(Dispatcher.java:109)
	at com.google.common.eventbus.EventBus.post(EventBus.java:267)
	at io.github.dsheirer.module.decode.p25.P25TrafficChannelManager.processPhase1ChannelGrant(P25TrafficChannelManager.java:374)
	at io.github.dsheirer.module.decode.p25.P25TrafficChannelManager.processChannelGrant(P25TrafficChannelManager.java:253)
	at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderState.processChannelGrant(P25P1DecoderState.java:333)
	at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderState.processTSBKGroupVoiceChannelGrant(P25P1DecoderState.java:1745)
	at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderState.processTSBK(P25P1DecoderState.java:1110)
	at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderState.receive(P25P1DecoderState.java:308)
	at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderState.receive(P25P1DecoderState.java:153)
	at io.github.dsheirer.sample.Broadcaster.broadcast(Broadcaster.java:124)
	at io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:48)
	at io.github.dsheirer.module.decode.Decoder$MessageDistributor.receive(Decoder.java:97)
	at io.github.dsheirer.module.decode.Decoder$MessageDistributor.receive(Decoder.java:90)
	at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageProcessor.receive(P25P1MessageProcessor.java:92)
[... cut ...]
	at io.github.dsheirer.util.Dispatcher$ProcessorWithHeartbeat.run(Dispatcher.java:224)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.base/java.util.concurrent.FutureTask.runAndReset(Unknown Source)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.base/java.lang.Thread.run(Unknown Source)`

MikeRAus, 2023-04-23:
In 0.5.0 and 0.5.2 I can clear hungup calls (STATUS has black background) by closing that Channel and restarting it.

jammcpe, 2023-11-13:
I am still getting this issue in 0.6.0 beta 5 on a p25 phase 2 trunked system I have all logs saved from the most recent time this happended and can provide any helpful info. I am able to resolve it by restarting the channel under the playlist editor. I have at points had it happen with multiple channels leading to 2 or 3 (6 total time slots) frequencies being left in teardown.

DSheirer, 2024-02-10:
Issue resolved with #1802.

If this issue is still happening with version 0.6.1 or later, there is a new menu option (File >> Reports >> Create Diagnostic that will capture the information I need to diagnose what's causing your channels to be stuck in TEARDOWN state.  Please create a new issue and attach that report to the issue.

closed 2024-02-10 by DSheirer