DSheirer/sdrtrunk #1523: Audio cuts after AMBE codec loaded

hans109h, 2023-04-14:
**sdrtrunk Version**
6.0-Alpha5

**Describe the bug**
Program starts normally and outputs stereo audio, then after some time both audio channels stop and no more audio is heard.  The log at this time reports:

2023-04-14 14:40:31.923 INFO  i.g.d.a.c.m.AmbeAudioModule - AMBE CODEC successfully loaded - P25-2/DMR/NXDN audio will be available  [284MB/568MB 50%]

This is reported a good 5 minutes after the IMBE codec reports as successfully loading.

**To Reproduce**
Launch program, wait for AMBE CODEC to successfully load.

**Expected behavior**
I would expect the codec to load much earlier, and this wouldn't be an issue that would affect existing channels.

**Screenshots**

**Application Log**
2023-04-14 14:34:10.243 INFO  i.g.d.icon.IconModel - loading icons file [/home/SDRTrunk/settings/icons.xml]  [24MB/1GB 2%]
2023-04-14 14:34:10.462 INFO  i.g.d.log.ApplicationLog - Application Log File: /home/SDRTrunk/logs/sdrtrunk_app.log  [51MB/1GB 5%]
2023-04-14 14:34:10.472 INFO  i.g.d.log.ApplicationLog - SDRTrunk Version  : 0.6.0-alpha5  [51MB/1GB 5%]
2023-04-14 14:34:10.472 INFO  i.g.d.log.ApplicationLog - Gradle Version    : Gradle 7.6  [51MB/1GB 5%]
2023-04-14 14:34:10.472 INFO  i.g.d.log.ApplicationLog - Build Timestamp   : 2023-04-14T12:54:48.202-0500  [51MB/1GB 5%]
2023-04-14 14:34:10.472 INFO  i.g.d.log.ApplicationLog - Build-JDK         : 19.0.2 (BellSoft 19.0.2+9  [51MB/1GB 5%]
2023-04-14 14:34:10.472 INFO  i.g.d.log.ApplicationLog - Build OS          : Linux (amd64 6.2.0-20-generic  [51MB/1GB 5%]
2023-04-14 14:34:10.472 INFO  i.g.d.log.ApplicationLog -   [51MB/1GB 5%]
2023-04-14 14:34:10.472 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [51MB/1GB 5%]
2023-04-14 14:34:10.473 INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application ***  [51MB/1GB 5%]
2023-04-14 14:34:10.473 INFO  i.g.d.log.ApplicationLog - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [51MB/1GB 5%]
2023-04-14 14:34:10.473 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [51MB/1GB 5%]
2023-04-14 14:34:10.473 INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%]  [51MB/1GB 5%]
2023-04-14 14:34:10.473 INFO  i.g.d.log.ApplicationLog - Host OS Name:          Linux  [51MB/1GB 5%]
2023-04-14 14:34:10.473 INFO  i.g.d.log.ApplicationLog - Host OS Arch:          amd64  [51MB/1GB 5%]
2023-04-14 14:34:10.473 INFO  i.g.d.log.ApplicationLog - Host OS Version:       6.2.0-20-generic  [51MB/1GB 5%]
2023-04-14 14:34:10.473 INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        16  [51MB/1GB 5%]
2023-04-14 14:34:10.474 INFO  i.g.d.log.ApplicationLog - Host Max Java Memory:  15 GB  [51MB/1GB 5%]
2023-04-14 14:34:10.474 INFO  i.g.d.log.ApplicationLog - Storage Directories:  [51MB/1GB 5%]
2023-04-14 14:34:10.474 INFO  i.g.d.log.ApplicationLog -  Application Root: /home/SDRTrunk  [51MB/1GB 5%]
2023-04-14 14:34:10.474 INFO  i.g.d.log.ApplicationLog -  Application Log:  /home/SDRTrunk/logs  [51MB/1GB 5%]
2023-04-14 14:34:10.474 INFO  i.g.d.log.ApplicationLog -  Event Log:        /home/SDRTrunk/event_logs  [51MB/1GB 5%]
2023-04-14 14:34:10.474 INFO  i.g.d.log.ApplicationLog -  Playlist:         /home/SDRTrunk/playlist  [51MB/1GB 5%]
2023-04-14 14:34:10.474 INFO  i.g.d.log.ApplicationLog -  Recordings:       /home/SDRTrunk/recordings  [51MB/1GB 5%]
2023-04-14 14:34:10.508 INFO  i.g.d.util.ThreadPool - Application thread pool created SCHEDULED and CACHED executors threads  [55MB/1GB 5%]
2023-04-14 14:34:10.508 INFO  i.g.dsheirer.gui.SDRTrunk - Home path: /home/SDRTrunk  [55MB/1GB 5%]
2023-04-14 14:34:10.509 INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [/home/SDRTrunk/SDRTrunk.properties]  [55MB/1GB 5%]
2023-04-14 14:34:10.509 INFO  i.g.d.p.SystemProperties - SystemProperties - application properties loaded [/home/SDRTrunk/SDRTrunk.properties]  [55MB/1GB 5%]
2023-04-14 14:34:10.549 INFO  i.g.d.s.t.m.TunerManager - Discovering tuners ...  [19MB/1GB 1%]
2023-04-14 14:34:10.554 INFO  i.g.d.s.t.m.TunerManager - LibUsb API Version: 1.0.262  [20MB/1GB 2%]
2023-04-14 14:34:10.554 INFO  i.g.d.s.t.m.TunerManager - LibUsb Version: 1.0.22.11312  [20MB/1GB 2%]
2023-04-14 14:34:10.556 INFO  i.g.d.s.t.m.TunerManager - LibUsb - discovered [17] potential usb devices  [20MB/1GB 2%]
2023-04-14 14:34:10.559 INFO  i.g.d.s.t.m.TunerManager - Discovered tuner at USB Bus [1] Port [10.4.2] Tuner Class [RTL-2832]  [20MB/1GB 2%]
2023-04-14 14:34:10.560 INFO  i.g.d.s.t.m.TunerManager - Discovered tuner at USB Bus [1] Port [10.3] Tuner Class [RTL-2832]  [20MB/1GB 2%]
2023-04-14 14:34:10.560 INFO  i.g.d.s.t.m.TunerManager - Discovered tuner at USB Bus [1] Port [10.1] Tuner Class [RTL-2832]  [20MB/1GB 2%]
2023-04-14 14:34:10.561 INFO  i.g.d.s.t.m.TunerManager - Tuner: USB Tuner - RTL-2832 USB Bus:1 Port:10.4.2 - Added / Starting ...  [21MB/1GB 2%]
2023-04-14 14:34:11.688 INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [18MB/136MB 13%]
2023-04-14 14:34:11.910 INFO  i.g.d.s.t.m.TunerManager - Tuner: USB Tuner - RTL-2832 USB Bus:1 Port:10.3 - Added / Starting ...  [18MB/136MB 13%]
2023-04-14 14:34:12.996 INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [19MB/136MB 14%]
2023-04-14 14:34:13.209 INFO  i.g.d.s.t.m.TunerManager - Tuner: USB Tuner - RTL-2832 USB Bus:1 Port:10.1 - Added / Starting ...  [19MB/136MB 14%]
2023-04-14 14:34:14.301 INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [20MB/136MB 15%]
2023-04-14 14:34:14.516 INFO  i.g.d.s.t.m.TunerManager - LibUsb - starting HotPlug detection  [21MB/136MB 15%]
2023-04-14 14:34:14.528 INFO  i.g.d.s.t.m.TunerManager - Tuner plug-in detected at USB Bus [1] Port [2] Tuner Class [RTL-2832]  [21MB/136MB 16%]
2023-04-14 14:34:14.529 INFO  i.g.d.s.t.m.TunerManager - Tuner plug-in detected at USB Bus [1] Port [3] Tuner Class [RTL-2832]  [21MB/136MB 16%]
2023-04-14 14:34:14.530 INFO  i.g.d.s.t.m.TunerManager - Tuner plug-in detected at USB Bus [1] Port [1] Tuner Class [RTL-2832]  [21MB/136MB 16%]
2023-04-14 14:34:14.538 INFO  i.g.d.s.t.s.api.SDRplay - Loading API Library from default install path: /usr/local/lib/libsdrplay_api.so  [21MB/136MB 16%]
2023-04-14 14:34:14.538 ERROR i.g.d.s.t.s.api.SDRplay - Unable to load SDRplay API library from default install path.  Loading from java system library path  [21MB/136MB 16%]
2023-04-14 14:34:14.538 WARN  i.g.d.s.t.s.api.SDRplay - SDRPlay API library not found/installed on this system.  Ensure the API is installed either in the default install location or the install location is included in the 'java.library.path' JVM property contains path to the library file [libsdrplay_api.so].  Current library path property contents: /usr/local/lib  [21MB/136MB 16%]
2023-04-14 14:34:14.539 INFO  i.g.d.s.t.s.api.SDRplay - API library was not loaded  [21MB/136MB 16%]
2023-04-14 14:34:14.539 INFO  i.g.d.s.t.s.api.SDRplay - API library is not available.  [21MB/136MB 16%]
2023-04-14 14:34:14.542 INFO  i.g.d.s.SettingsManager - SettingsManager - loading settings file [/home/SDRTrunk/settings/settings.xml]  [21MB/136MB 16%]
2023-04-14 14:34:15.001 INFO  i.g.d.p.PlaylistManager - Loading playlist [/home/SDRTrunk/playlist_backup_2/backups_alpha7/default_clone.xml]  [47MB/136MB 34%]
2023-04-14 14:34:15.208 INFO  i.g.dsheirer.gui.SDRTrunk - starting main application gui  [39MB/136MB 29%]
2023-04-14 14:34:15.234 INFO  i.g.d.a.b.AudioStreamingBroadcaster - [Dane County] status: Connected  [43MB/136MB 32%]
2023-04-14 14:34:18.183 INFO  i.g.d.c.c.ChannelAutoStartFrame - Starting [1] channels now - user invoked  [44MB/136MB 32%]
2023-04-14 14:34:18.193 INFO  i.g.d.s.t.m.TunerManager - Unable to source channel [172500000] from preferred tuner [RTL2832 SDR/R820T 00000104] - searching for another tuner  [39MB/136MB 29%]
2023-04-14 14:34:18.267 INFO  i.g.d.a.c.m.JmbeAudioModule - Loading JMBE library from [/home/SDRTrunk/jmbe/jmbe-1.0.9.jar]  [43MB/136MB 31%]
2023-04-14 14:34:18.270 INFO  i.g.d.a.c.m.JmbeAudioModule - JMBE audio conversion library loaded: JMBE Audio Conversion Library v1.0.9  [43MB/136MB 31%]
[... cut ...]
2023-04-14 14:34:55.578 INFO  i.g.d.s.t.m.FrequencyErrorCorrectionManager - Auto-Correcting Tuner PPM to [-1.5]  [118MB/240MB 49%]
2023-04-14 14:40:13.793 INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [237MB/352MB 67%]
2023-04-14 14:40:31.923 INFO  i.g.d.a.c.m.AmbeAudioModule - AMBE CODEC successfully loaded - P25-2/DMR/NXDN audio will be available  [284MB/568MB 50%]

**Desktop (optional - complete the following information):**
 - OS: Linux
 - CPU Cores: 16
 - RAM: Lots

**Additional context**
Add any other context about the problem here.

DSheirer, 2023-04-15:
The AMBE codec gets loaded on-demand, whenever it's needed.  It won't cause audio to cutout.  

I suspect your audio drop out may be caused by some other issue.

On the channel that's producing the audio that cuts out, are you still seeing a strong signal in the channel view?

hans109h, 2023-04-15:
Yes, everything else operates as normal, calls come in and are reported as being output on left and right, just no audio.

Usually both left and right cut at the same time, but one time left cut about a minute before right.   I'll try to do some more troubleshooting.

hans109h, 2023-04-15:
Working back, the last good configuration was 6.0alpha4, after that audio cuts out shortly after starting, sometimes as soon as 15 seconds, others about 5 minutes.

hans109h, 2023-04-24:
Changed the title as it is not longer corrolated with the codec.  Still having the same issue where after sometime audio will just cut out on one channel or the other, then both.  I can start another channel (such as NBFM) and it will play the audio.

I've confirmed that the audio (that isn't being played in the software) is still being feed to my icecast stream.

DSheirer, 2023-05-21:
@hans109h Can you please test against the latest Version 0.6.0 Alpha 6 release to see if this is still an issue?

hans109h, 2023-05-22:
@DSheirer this is still an issue.  I built the latest and it still cuts out, and just to be sure I downloaded your latest pre-release build and it occurs.

I've been able to "reset" the audio when it occurs by going into preferences and changing the audio output from stereo to mono or vice-versa and then I get audio again for a bit, but it eventually stops on a per channel basis.  Nothing in the sdrtrunk logs or system logs have been helpful.

hans109h, 2023-07-08:
So it's seaming that this may only be a related issue to sdrtrunk, as I can only reproduce it when 'audio output device' is set to default stereo or mono.  I need to do additional testing to see if that is because of the system default selected or how sdrtrunk is using default.

So far I am able to test a built in sound card line out--this has been my default since the beginning and for whatever reason after 6alpha4 this bug is present.  I changed today to stereo output through an HDMI connection and the bug is not present, however even though this device (and the setting in sdrtrunk) is stereo, I only get output on the left channel.

If I set my system (ubuntu 23.04) default audio device to the HDMI audio out, and then use the default stereo setting in sdrtrunk it does produce audio on both the left and right channels.

When my system audio is set to sound card line out, but I change to use the HDMI audio device in sdrtrunk I am seeing this error:
ERROR i.g.d.a.p.AudioOutput - Couldn't obtain audio source data line for audio output - mixer [PCH [plughw:0,7]]

hans109h, 2023-07-31:
This appeared to be an issue with PulseAudio on Ubuntu 23.04.  I would suspect that a fresh install of 23.04, which ships with pipewire and no PA, would not result in this problem.   

I purged PA and no longer have this issue.  Through pipewire, as long as SDRTrunk is set to default output, I can direct audio to an HDMI sink and it works in stereo.  If I try to use this sink directly in sdrtrunk I only get mono.  I suspect this is an issue with how java is connected/not connect to pipewire.

I'm closing this issue.  If I figure out anything new with regards to mono vs stereo within SDRTrunk I will open a new ticket.

closed 2023-07-31 by hans109h