DSheirer/sdrtrunk #1526: SDRTrunk is less sensitive than DSDPlus and SDRSharp + reducing the bandwidth for DMR and NFM

noamlivne, 2023-04-18:
Comparing between the same signal/antenna/dongle, it seems that SDRTrunk is less sensitive than DSDPlus Fastlane and SDRSharp.
The signal is (almost) not received in SDRTrunk, but is strong enough in DSDPlus and SDRSharp.
As if the noise floor is very high in SDRTrunk and it is less able to detect the signal from the noise.
In addition, it would help if the DMR bandwidth will be reduced to about 8KHz and the NFM would be reduced to 8 or even less.

reppiZ, 2023-04-20:
On the "Tuners" tab underneath the FFT display, have you messed with the gain control settings for the tuner?

DSheirer, 2023-05-07:
@noamlivne when you ran these tests, were you using the Polyphase (default) or Heterodyne channelizers, or did you test with both?  

The heterodyne channelizer uses half-band decimators which add ~3db of gain per decimation stage and this is probably closer to the SDRSharp implementation. The polyphase channelizer is more computationally efficient but doesn't have any of the gain-adding benefits.

This issue requests to change the channel bandwidth for the DMR & NBFM channels.  The NBFM decoder allows you to select between 15 & 25 kHz.  Do you have local FM channels that are using 8 kHz bandwidth?  I can add support for that bandwidth if there's use case for it.

DMR is designed to occupy a 12.5 kHz channel and the baseband I/Q filter is set for 13.0 kHz with roll off starting at 10.2 kHz.  You can test changing that filter to an 8 kHz channel bandwidth -- the code is in the ```DMRDecoder``` class in the ```getBasebandFilter()``` method.

ah1102, 2023-05-08:
I have two P25P1 sites side by side. When you turn on two of them simultaneously or sequentially, a drop in sensitivity is observed.

The peak was -42.9, but fell to -55.1

[image]
[image]

noamlivne, 2023-05-09:
@DSheirer 
Hi Denny,
Thank you very much for your support and suggestions.
Unfortunately, my friend and I did not notice any difference (graphical or reception-wise) when changing between Polyphase and Heterodyne channelizers.
DSDPlus manages to pick up the weak signals and decode them, but SDRTrunk did not pick them up or decode them.

I asked my friend that knows how to make changes in the code, to use smaller values with NFM and DMR, but he said that he did not notice any improvement.
I offered that narrowing because I manage to pick up weaker NFM and DMR (and P25 control channels, less CRC errors) with narrower bandwidth, such as 7.6KHz, instead of 9.5 or 12.5. (With NXDN48 I even pick up the very weak signals with 4KHz filter). At least this is my personal experience.

Thank you again for every future improvement.

closed 2023-11-14 by DSheirer