DSheirer/sdrtrunk #1527: SDRTrunk Broadcasting feed

hccr717, 2023-04-18:
So i am trying to provide a audio feed to broacasting and when i set it up and enable it all it what to do is connect and disconnect non stop please help me. Im new at this stuff

reppiZ, 2023-04-20:
What broadcasting service are you trying to use? Icecast, Broadcastify?

hccr717, 2023-04-21:
Broadcastify

hccr717, 2023-05-01:
guest no one knows

DSheirer, 2023-05-02:
If it's trying to connect, failing and then disconnecting, then it sounds like you might have some sort of networking issue.  The software is trying to connect to the broadcastify server and failing.

There may be an error listed in the application log.  Can you post here and attach the application log?

hccr717, 2023-05-02:
what logs do i need to look at? new at this

reppiz, 2023-05-02:
You doing this over Wi-Fi or Ethernet? I used to use Broadcastify before making my own icecast server and never had any issues with using them. Definitely sounds like a network problem and less of a problem
With SDRTrunk.

hccr717, 2023-05-02:
how do u do it over icecast

reppiz, 2023-05-02:
You need to have the icecast server running on a machine, then you put in the IP, mount point, and password of the icecast server into the fields on SDRTrunk.

hccr717, 2023-05-02:
ok how

reppiz, 2023-05-02:
Little too complicated to write it out here. I'd be happy to help via a voice chat over Discord or something similar. Or you can try to Google it yourself.

DSheirer, 2023-05-07:
Closing.  This appears to be more of a user-help request and not a bug/anomaly with sdrtrunk clode.

closed 2023-05-07 by DSheirer