DSheirer/sdrtrunk #1535: Forced disable DMR trunk detection.

Dmedia3412, 2023-05-01:
Please implement this feature. For encrypted DMR calls in simple conventional systems, the calls are displayed incorrectly.

DSheirer, 2023-05-07:
Could you please explain this feature request?  I don't understand what you're asking for?

Dmedia3412, 2023-05-07:
I would like the program not to determine the type or presence of any trunk itself. So that it could be turned off with a switch in the channel settings and the program forcibly considered that this was a simple conventional signal. This is because Hytera's simple conventional system is defined as CON+. This spoils the entire display of RID, TGID, TS information. Perhaps in this way the program will display the correct information? I have already spoken about mistakes several times, but apparently no one needs it except me. I have already tried all versions of the program, but they all incorrectly determine that a simple conventional system is CON +.

DSheirer, 2023-05-15:
@Dmedia3412 could you please generate a DMR bitstream (*.bits) recording for this system and either send it to me or attach it to this issue?  I'd like to fix/update any message(s) that are being mis-identified as CON+.

Dmedia3412, 2023-05-15:
I can't find your email contacts. I don't want to give to everyone.

closed 2023-05-16 by Dmedia3412