DSheirer/sdrtrunk #1542: Error building package (macOS x64)

ah1102, 2023-05-07:
When trying to build the current program code, an error is generated

```
Deprecated Gradle features were used in this build, making it incompatible with Gradle 8.0.

BUILD FAILED in 11s
```

DSheirer, 2023-05-08:
Please use Gradle 7.6.1 and JDK 19 until the codebase is updated.

I'm awaiting the release of Gradle 8.2.0 to update to JDK20.

closed 2023-05-08 by DSheirer

ah1102, 2023-05-08:
You write in the manual that **"Java JDK 19+ with integrated Open JavaFX"** is required

https://github.com/DSheirer/sdrtrunk/wiki/Developing

Hard to find an answer:

> Thanks for your help Carl. I had a lot of false starts, but I finally got it to build. For anyone else wanting to build this, the solution is to download the *full* installer, not the "standard" one. On, https://bell-sw.com/pages/downloads/ click the dropdown where it says "Standard JDK" and you will see the "Full JDK" option.
> 
> https://groups.google.com/g/sdrtrunk/c/RpPZQUCtd0w/m/sgNZPKMMBgAJ  

It would be nice to edit the manual by writing in brackets next to the package you recommend: **Bellsoft Liberica OpenJDK (Package Type: "Full JDK")**

[image]

ah1102, 2023-05-08:
The same thing happens with Windows OS. I have seen JDK 19 and Gradle 7.6 loaded the first time the script is run `gradlew runtimeZipWindows`. But the same errors appear.

```
C:\Users\ah110\Desktop\sdrtrunk-master>gradlew runtimeZipWindows

> Task :compileJava
warning: using incubating module(s): jdk.incubator.vector
C:\Users\ah110\Desktop\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\action\AliasAction.java:30: error: package javafx.beans does not exist
import javafx.beans.Observable;
                   ^
C:\Users\ah110\Desktop\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\action\AliasAction.java:31: error: package javafx.beans.property does not exist
import javafx.beans.property.SimpleStringProperty;
                            ^
C:\Users\ah110\Desktop\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\action\AliasAction.java:32: error: package javafx.util does not exist
import javafx.util.Callback;
                  ^
C:\Users\ah110\Desktop\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\action\AliasAction.java:47: error: cannot find symbol
    private SimpleStringProperty mValueProperty = new SimpleStringProperty();
            ^
  symbol:   class SimpleStringProperty
  location: class AliasAction
C:\Users\ah110\Desktop\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\action\AliasAction.java:57: error: cannot find symbol
    public SimpleStringProperty valueProperty()
           ^
  symbol:   class SimpleStringProperty
  location: class AliasAction
C:\Users\ah110\Desktop\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\Alias.java:33: error: package javafx.beans does not exist
import javafx.beans.Observable;
                   ^
C:\Users\ah110\Desktop\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\Alias.java:34: error: package javafx.beans.binding does not exist
import javafx.beans.binding.Bindings;
                           ^
C:\Users\ah110\Desktop\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\Alias.java:35: error: package javafx.beans.binding does not exist
import javafx.beans.binding.BooleanBinding;
                           ^
C:\Users\ah110\Desktop\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\Alias.java:36: error: package javafx.beans.property does not exist
import javafx.beans.property.BooleanProperty;
                            ^
C:\Users\ah110\Desktop\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\Alias.java:37: error: package javafx.beans.property does not exist
import javafx.beans.property.IntegerProperty;
                            ^
C:\Users\ah110\Desktop\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\Alias.java:38: error: package javafx.beans.property does not exist
import javafx.beans.property.SimpleBooleanProperty;
                            ^
C:\Users\ah110\Desktop\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\Alias.java:39: error: package javafx.beans.property does not exist
import javafx.beans.property.SimpleIntegerProperty;
                            ^
C:\Users\ah110\Desktop\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\Alias.java:40: error: package javafx.beans.property does not exist
import javafx.beans.property.SimpleStringProperty;
                            ^
C:\Users\ah110\Desktop\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\Alias.java:41: error: package javafx.beans.property does not exist
import javafx.beans.property.StringProperty;
                            ^
C:\Users\ah110\Desktop\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\Alias.java:42: error: package javafx.collections does not exist
import javafx.collections.FXCollections;
                         ^
C:\Users\ah110\Desktop\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\Alias.java:43: error: package javafx.collections does not exist
import javafx.collections.ObservableList;
                         ^
C:\Users\ah110\Desktop\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\Alias.java:44: error: package javafx.collections.transformation does not exist
import javafx.collections.transformation.FilteredList;
                                        ^
C:\Users\ah110\Desktop\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\Alias.java:45: error: package javafx.util does not exist
import javafx.util.Callback;
                  ^
C:\Users\ah110\Desktop\sdrtrunk-master\src\main\java\io\github\dsheirer\alias\action\AliasAction.java:93: error: cannot find symbol
    public static Callback<AliasAction,Observable[]> extractor()
                  ^
  symbol:   class Callback
  location: class AliasAction
[... 365 lines cut ...]

> Task :compileJava FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':compileJava'.
> Compilation failed; see the compiler error output for details.

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

Deprecated Gradle features were used in this build, making it incompatible with Gradle 8.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.

See https://docs.gradle.org/7.6/userguide/command_line_interface.html#sec:command_line_warnings

BUILD FAILED in 32s
1 actionable task: 1 executed
```

ah1102, 2023-05-08:
Probably the `./gradlew runtimeZip` script doesn't download the full JDK. Which doesn't have JavaFX. Or accessing the wrong JDK