DSheirer/sdrtrunk #1546: Java display white out

rkcheung, 2023-05-10:
**sdrtrunk Version**
v0.6.0-alpha 5 and v0.5.3 final

**Describe the bug**
After some hours of running, the Java app display will white out and controls become inaccessible.

**Desktop:**
 - OS: Ventura 13.0
 - CPU Cores: Apple M1
 - RAM: 8 GB
 
[image]

DSheirer, 2023-05-15:
@rkcheung there's not enough detail for me to diagnose what's going on.

My best guess is that this may be related to a display power-saving mode that shuts down the display after a period of inactivity and it may be failing to repaint the screen.  If that's the case, this would be a java bug, not a bug in sdrtrunk application.

Can you please recreate the scenario that produces this white screen and then send me the application log so that I can check if there were any errors logged.  Otherwise, there's not much I can do with this issue.

rkcheung, 2023-05-29:
This issue has not re-occured recently so I'll close it.

closed 2023-05-29 by rkcheung